zlgopen/awtk #557: 能否将不同目标平台的编译输出的库文件放在不同的目录？

ufbycd, 2021-05-31:
编译不同的目标平台（scons、scons LINUX_FB=true）时，目录由于库文件目录没有按平台分开，切换平台时就要执行消除再重新编译；如果用到的awtk相关的外部控件比较多的话，编译量太大，过程太繁琐。
    如果能将不同平台输出的库文件放在不同的目录，将方便很多。

xianjimli, 2021-06-02:
我看了一下，不太好弄。

ufbycd, 2021-06-07:
awtk及awtk-linux-fb工程在切换平台时不需要重复编译，所以主要解决awtk-mvvm这些工程的构建即可。

我尝试改了下，似乎可以，改动如下：

```diff
diff --git a/scripts/app_helper_base.py b/scripts/app_helper_base.py
index bb290a02d..401ceb91b 100644
--- a/scripts/app_helper_base.py
+++ b/scripts/app_helper_base.py
@@ -160,6 +160,18 @@ class AppHelperBase:
         if plat == PLATFORM:
             self.APP_LINKFLAGS += APP_LINKFLAGS
         return self
+        
+    def SConscript(self, SConscript, SConscriptFiles):
+        if len(self.BUILD_DIR) == 0:
+            SConscript(SConscriptFiles)
+        else:
+            for sc in SConscriptFiles:
+                dir = os.path.dirname(sc)
+                if len(dir) == 0:
+                    build_dir = self.BUILD_DIR
+                else:
+                    build_dir = os.path.join(self.BUILD_DIR, dir)
+                SConscript(sc, variant_dir=build_dir, duplicate=False)
 
     def __init__(self, ARGUMENTS):
         APP_ROOT = os.path.normpath(os.getcwd())
@@ -179,8 +191,11 @@ class AppHelperBase:
         self.AWTK_CFLAGS = self.awtk.CFLAGS
         self.AWTK_CCFLAGS = self.awtk.CCFLAGS
         self.APP_ROOT = APP_ROOT
-        self.APP_BIN_DIR = os.path.join(APP_ROOT, 'bin')
-        self.APP_LIB_DIR = os.path.join(APP_ROOT, 'lib')
+        self.BUILD_DIR = ARGUMENTS.get('BUILD_DIR', '')
+        self.BIN_DIR = os.path.join(self.BUILD_DIR, 'bin')
+        self.LIB_DIR = os.path.join(self.BUILD_DIR, 'lib')
+        self.APP_BIN_DIR = os.path.join(APP_ROOT, self.BIN_DIR)
+        self.APP_LIB_DIR = os.path.join(APP_ROOT, self.LIB_DIR)
         self.APP_SRC = os.path.join(APP_ROOT, 'src')
         self.APP_RES = os.path.join(APP_ROOT, 'res')
         self.APP_LIBS = []
@@ -279,7 +294,8 @@ class AppHelperBase:
 
         for iter in self.DEPENDS_LIBS:
             for so in iter['shared_libs']:
-                self.awtk.copySharedLib(iter['root'], self.APP_BIN_DIR, so)
+                src = os.path.join(iter['root'], self.BUILD_DIR)
+                self.awtk.copySharedLib(src, self.APP_BIN_DIR, so)
 
     def genIdlAndDef(self):
         if self.DEF_FILE:
@@ -311,6 +327,7 @@ class AppHelperBase:
         print('  THEME="default"')
         print('  SHARED=true')
         print('  LINUX_FB=false')
+        print('  BUILD_DIR=""')
         sys.exit(0)
 
     def parseArgs(self, awtk, ARGUMENTS):
@@ -506,8 +523,8 @@ class AppHelperBase:
                     LIBPATH = LIBPATH + [join_path(iter['root'], f)]
             else:
                 if self.isBuildShared():
-                    LIBPATH += [join_path(iter['root'], 'bin')]
-                LIBPATH += [join_path(iter['root'], 'lib')]
+                    LIBPATH += [join_path(iter['root'], self.BIN_DIR)]
+                LIBPATH += [join_path(iter['root'], self.LIB_DIR)]
         LIBS = self.APP_LIBS + LIBS
 
         self.prepare()
```

SConstruct文件里的SConscript函数要改为使用app_helper_base.py里提供的SConscript提供的函数，如：
```python
SConscriptFiles = ['src/SConscript', 'demos/SConscript', 'tests/SConscript']
helper.SConscript(SConscript, SConscriptFiles)
```

然后在建构时通过`BUILD_DIR`指定编译输出目录即可，如：
```
scons LINUX_FB=true BUILD_DIR=arm
```

[0001-Add-parameter-BUILD_DIR-to-scons-to-specify-build-di.patch.txt](https://github.com/zlgopen/awtk/files/6607722/0001-Add-parameter-BUILD_DIR-to-scons-to-specify-build-di.patch.txt)

xianjimli, 2021-06-07:
谢谢。还有release等配套脚本，你自己先这样用吧。

ufbycd, 2021-06-09:
修改之后发现，每次构建都会编译编译位于`awtk/3rd/gtest/googletest/src`目录下的`getst-all.cc`文件。

于是，我把`tests/SConscript`文件改成这样，不知是否妥当？

```diff
-SOURCES = [
- os.path.join(GTEST_ROOT, 'src/gtest-all.cc'),
-] + Glob('*.cc') + Glob('*.c')
+gtest = 'gtest-all.cc'
+if not os.path.exists(gtest):
+    src = os.path.join(GTEST_ROOT, 'src', gtest)
+    print('link {} -> {}'.format(src, gtest))
+    os.link(src, gtest)
 
-env.Program(os.path.join(BIN_DIR, 'runTest'), SOURCES);
+SOURCES = Glob('*.cc') + Glob('*.c')
+
+env.Program(os.path.join(BIN_DIR, 'runTest'), SOURCES)
```

ufbycd, 2021-06-11:
发现用`scons`内置的`Copy Factory`来处理`gtest-all.cc`文件要好很多：
```python
gtest = 'gtest-all.cc'
src = os.path.join(GTEST_ROOT, 'src', gtest)
env.Command(gtest, src, Copy("$TARGET", "$SOURCE"))
SOURCES = Glob('*.cc') + Glob('*.c')
```

ufbycd, 2021-06-11:
又有新发现：scons专有脚本文件里的`SConscript` `DefaultEnvironment`之类的东西的命名空间来至模块`SCons.Script`，那么要想在专有脚本文件之外使用这些东西，只需导入`SCons.Script`模块即可。

于是，`app_helper_base.py`里的`SConscript`函数就可以改成这样：
```python
from SCons import Script

def SConscript(self, SConscriptFiles):
    if not self.BUILD_DIR:
        Script.SConscript(SConscriptFiles)
    else:
        for sc in SConscriptFiles:
            dir = os.path.dirname(sc)
            build_dir = os.path.join(self.BUILD_DIR, dir)
            Script.SConscript(sc, variant_dir=build_dir, duplicate=False)
```

xianjimli, 2021-06-12:
试试了一下，挺好用！没想到几行代码就搞定了。等我在不同平台测试一下，再合并进去。

xianjimli, 2021-06-27:
不好意思，最近太忙，没有合并。结果忘记怎么弄的了，你能把最新的补丁发给我吗？谢谢

ufbycd, 2021-06-28:
[add-parameter-BUILD_DIR-to-build-script.patch.txt](https://github.com/zlgopen/awtk/files/6722690/add-parameter-BUILD_DIR-to-build-script.patch.txt)

xianjimli, 2021-06-28:
合并了，非常感谢。

closed 2021-06-29 by ufbycd