zlgopen/awtk #558: rlottie widget for Toolkit AnyWhere

ajsb85, 2021-06-02:
Allows to use rlottie animations in Toolkit AnyWhere

You can use [Samsung/rlottie](https://github.com/Samsung/rlottie) library with C-api header

Best regards.

Ref:
https://github.com/Samsung/rlottie/issues/484

xianjimli, 2021-06-03:
OK, thanks.

ajsb85, 2021-06-04:
Dear @xianjimli 

To download the UI Designer is very difficult to me because I have no Chinese nor Asian phone number.
Baidu or the AWTK云平台 needs as mandatory those numbers.

Can you send to me by Wetransfer maybe?

a.salas@ieee.org

Regards.

soyzhc, 2021-06-05:
@ajsb85 

You can use a international mobile number to register an account with awtk.zlg.cn.
Select the mobile area code in the first registration step.

Please try again. and let me know if there is any exception during the registration process.

ajsb85, 2021-06-05:
I can record the video of my frustrations
After several try to get the SMS finally the platform says that they can activate the account

My phone number: +34697703290
My email: alexander.salas@gmail.com

soyzhc, 2021-06-05:
I'll have my colleagues in charge follow up on this next working day

xianjimli, 2021-06-05:
Lottie works on AWTK now：https://github.com/zlgopen/awtk-widget-rlottie

ajsb85, 2021-06-05:
Wow!

Very impressive. Kudos to you @xianjimli 

Help me to get the GUI Editor and the Evaluation board.
To start helping you!

Which one is the best evaluation board?
Maybe this?

[image]

https://it.aliexpress.com/item/1005001794778646.html?spm=a2g0o.cart.0.0.6cc53c00OqoSrY&mp=1

closed 2021-06-05 by ajsb85

ajsb85, 2021-06-05:
Hi, @soyzhc 

I am enrolled finally 😃 

[image]

xianjimli, 2021-06-05:
😃