zlgopen/awtk #559: 阿拉伯语字符串中加入换行符后，通过label控件显示出来的内容出现了镜像翻转的问题

hantianheng, 2021-06-08:
你好，
我在使用label控件去展示阿拉伯语的信息时在一些特殊情况下需要对阿拉伯语的字符串进行换行处理，但是我发现经过换行处理之后的阿拉伯语字符串再通过label控件显示时出现了镜像翻转的问题。
例如：
我在strings.xml中加入下面这段内容
[image]

然后通过label控件将其显示出来，显示出来的结果变成了：

[image]

xianjimli, 2021-06-08:
用启用bidi吗？https://github.com/zlgopen/awtk/blob/master/docs/how_to_enable_bidi.md

你把文字贴到这里吧。

hantianheng, 2021-06-09:
> 用启用bidi吗？https://github.com/zlgopen/awtk/blob/master/docs/how_to_enable_bidi.md
> 
> 你把文字贴到这里吧。

启用了bidi
文字：
إنذار
مجموعة

使用方式：
<label name="F1_label" w="100%" y="-5" h="80"  tr_text="ALARM_INTERVAL" />

xianjimli, 2021-06-11:
改了。你有空更新代码试试。谢谢。

hantianheng, 2021-06-11:
好的，谢谢

hantianheng, 2021-06-15:
> 改了。你有空更新代码试试。谢谢。

你好，最新的AWTK 对于阿拉伯语的多行显示问题已经修复了。但是，这次的修改同时也引入了新的问题。

当我要显示的字符串是以“\n”或者“\r”开头时会触发断言。例如： “\n1234” , “12\n\n34” 
报错信息内容如下：

_bidi_log2vis:103 condition(bidi != NULL && str != NULL && size > 0) failed!
Assertion failed: (!"log2vis failed!"), function canvas_draw_text_bidi_in_rect, file src/base/canvas.c, line 1866.
Abort trap: 6_

查看了下代码发现是在label_paint_text_mlines()函数中对字符串的首个字符进行了判断

   while (line_parser_next(p) == RET_OK) {
    uint32_t size = 0;
    rect_t r = rect_init(x, y, w, font_size);

    if ((y + font_size) > bottom) {
      break;
    }

    for (size = 0; size < p->line_size; size++) {
      if (p->line[size] == '\r' || p->line[size] == '\n') {//如果符合条件则size = 0
          break;
      }
    }
    widget_draw_text_in_rect(widget, c, p->line, size, &r, FALSE);//如果size = 0 则后续的处理过程中就会触发断言。

我不是很清楚这样做的理由，但是考虑到要显示的文本的不确定性，这样做会很容易就触发断言。

我目前的做法是判断当size>0时，才会去执行widget_draw_text_in_rect
 if(size > 0)
    widget_draw_text_in_rect(widget, c, p->line, size, &r, FALSE);

xianjimli, 2021-06-15:
改了。你有空更新代码试试。谢谢。

hantianheng, 2021-06-15:
> 改了。你有空更新代码试试。谢谢。

好的，谢谢