zlgopen/awtk #560: scons构建系统优化建议

ufbycd, 2021-06-11:
1. 将编译输出文件放到源码文件之外的单独文件内，参见#557；
2. 使用`scons`内置的`Copy Factory`来复制`awtk`里的`gtest-all.cc`文件到APP源码内，参见[这里](https://github.com/zlgopen/awtk/issues/557#issuecomment-859351050)；
3. 使用`Copy Factory`来复制库文件，这样就不会每次甚至在`clean`时都复制文件了，`awtk_config.py`：
```python
from SCons import Script

def copySharedLib(src, dst, name):
    if OS_NAME == 'Darwin':
        lib_name = 'lib'+name+'.dylib'
    elif OS_NAME == 'Linux':
        lib_name = 'lib'+name+'.so'
    elif OS_NAME == 'Windows':
        lib_name = 'lib'+name+'.dll'
    else:
        print('not support ' + OS_NAME)
        return

    src = os.path.join(src, 'bin', lib_name)
    dst = os.path.join(dst, lib_name)
    src = os.path.normpath(src)
    dst = os.path.normpath(dst)
    if os.path.abspath(src) == os.path.abspath(dst):
        return
    #print(src + ' ==> ' + dst);
    Script.Command(dst, src, Script.Copy("$TARGET", "$SOURCE"))
```

4. 将release配套脚本整合到`scons`构建系统里并命令为'deploy'，通过 `scons deploy`来打包APP；（软件通常分为`debug`跟`release`两种版本，而目前`awtk`的release配套脚本就是用来打包，改名为`deploy`合适点）

xianjimli, 2021-06-11:
谢谢。引用自定义控件好弄吗。

ufbycd, 2021-06-11:
> 谢谢。引用自定义控件好弄吗。

引用自定义控件？没明白

xianjimli, 2021-06-11:
比如引用table view控件：https://github.com/zlgopen/awtk-widget-table-view

ufbycd, 2021-06-11:
可以的，控件里的`SConstruct `脚本里的`SConscript`调用改成这样即可：
```python
helper.SConscript(SConscriptFiles)
```

xianjimli, 2021-06-11:
生成的库和头文件在不同的目录，CPPPATH/LIBPATH如何设置呢

ufbycd, 2021-06-11:
在`app_helper_base.py`里处理，`self.BUILD_DIR`就等于`BUILD_DIR`参数的值

```python
def copyAwtkSharedLib(self):
    if self.TKC_ONLY:
        self.awtk.copySharedLib(self.AWTK_ROOT, self.APP_BIN_DIR, 'tkc')
    else:
        self.awtk.copySharedLib(self.AWTK_ROOT, self.APP_BIN_DIR, 'awtk')

    for iter in self.DEPENDS_LIBS:
        for so in iter['shared_libs']:
            src = os.path.join(iter['root'], self.BUILD_DIR)
            self.awtk.copySharedLib(src, self.APP_BIN_DIR, so)
```

ufbycd, 2021-06-11:
[app_helper_base.py.txt](https://github.com/zlgopen/awtk/files/6637689/app_helper_base.py.txt)

ufbycd, 2021-06-22:
发现一个问题：在修改awtk的src文件夹内的部分源码后构建系统没有重新生成awtk库（libawtk.so）；需要在修改源码后先构建一次，再手动删除bin/libawtk.so，然后再次构建才能生成跟所修改的源码对应的库文件。

大侠没遇到这个问题么？

ufbycd, 2021-06-22:
看了代码，原来是因为要用`--whole-archive`链接选项而没有将依赖库写到`LIBS`变量里；于是`scons`就不能根据`LIBS`来添加隐式依赖关系。

在网上找到这种[解决办法](https://blog.csdn.net/kevin_darkelf/article/details/80530026)，修改了下：
```diff
From fc567554b22d2d75582f8981b86a6e25403bcc37 Mon Sep 17 00:00:00 2001
From: ChenShiSheng <chenshisheng@reliachina.com>
Date: Tue, 22 Jun 2021 16:45:32 +0800
Subject: [PATCH] add the dependency of awtk library

---
 awtk_config.py | 3 ++-
 src/SConscript | 2 ++
 2 files changed, 4 insertions(+), 1 deletion(-)

diff --git a/awtk_config.py b/awtk_config.py
index c1bb853c4..dcb90da94 100644
--- a/awtk_config.py
+++ b/awtk_config.py
@@ -201,8 +201,9 @@ elif OS_NAME == 'Linux':
     OS_FLAGS = OS_FLAGS + ' -DWITH_64BIT_CPU '
 
   OS_LINKFLAGS=' -Wl,-rpath=./bin -Wl,-rpath=./ '
-  AWTK_DLL_DEPS_LIBS = NANOVG_BACKEND_LIBS + ['SDL2', 'glad'] + OS_LIBS
+  AWTK_DLL_DEPS_LIBS = AWTK_STATIC_LIBS + NANOVG_BACKEND_LIBS + ['SDL2', 'glad'] + OS_LIBS
   OS_WHOLE_ARCHIVE =' -Wl,--whole-archive -lawtk_global -lextwidgets -lwidgets -lbase -lgpinyin -lstreams -lconf_io -lhal -lcsv -lubjson -lcompressors -lfribidi -lmbedtls -lminiz -ltkc_static -llinebreak -Wl,--no-whole-archive'
+  OS_WHOLE_ARCHIVE += ' -Wl,--allow-multiple-definition '
 
 elif OS_NAME == 'Windows':
   if not os.path.exists(os.path.abspath(TK_BIN_DIR)) :
diff --git a/src/SConscript b/src/SConscript
index f5bfe11be..d3908ae15 100644
--- a/src/SConscript
+++ b/src/SConscript
@@ -110,6 +110,7 @@ OS_NAME = platform.system()
 TKC_LINKFLAGS = env['LINKFLAGS'];
 TKC_STATIC_LIBS = ['conf_io', 'hal', 'csv', 'ubjson', 'streams', 'tkc_static', 'compressors', 'mbedtls', 'miniz']
 TKC_LINK_ALL=' -Wl,--whole-archive -lstreams -lconf_io -lhal -lcsv -lubjson -ltkc_static -lcompressors -lmbedtls -lminiz -Wl,--no-whole-archive'
+TKC_LINK_ALL += ' -Wl,--allow-multiple-definition '
 
 if OS_NAME == 'Windows':
   if TOOLS_NAME == '' :
@@ -122,6 +123,7 @@ elif OS_NAME == 'Darwin':
   TKC_DEPS_LIBS= TKC_STATIC_LIBS
   TKC_LINKFLAGS +=' -all_load '
 elif OS_NAME == 'Linux':
+  TKC_DEPS_LIBS = TKC_STATIC_LIBS
   TKC_LINKFLAGS += TKC_LINK_ALL
 
 env.SharedLibrary(os.path.join(BIN_DIR, 'tkc'), [], LIBS=TKC_DEPS_LIBS, LINKFLAGS=TKC_LINKFLAGS); 
-- 
2.32.0
```
[0001-add-the-dependency-of-awtk-library.patch.txt](https://github.com/zlgopen/awtk/files/6692760/0001-add-the-dependency-of-awtk-library.patch.txt)

ufbycd, 2021-06-22:
不加`-Wl,--allow-multiple-definition`链接选项，似乎也没问题，`-l`指定的依赖库出现重复应该也没有问题。

所以重要还是要把依赖库写到`LIBS`变量里，让`scons`自动添加依赖关系。