zlgopen/awtk #562: Rive

ajsb85, 2021-06-14:
Dear @xianjimli 

Do you know about the Rive App?

Take a look this example by @jctoon 

This is an example of a state machine that you can use in Android, because the Rive Android runtime now supports state machines. #Bones #Icons #Characters #Android #StateMachine

https://rive.app/community/223-426-android-fun/

References:

https://github.com/rive-app/rive-cpp
https://github.com/rive-app/rive-tizen
https://github.com/rive-app/rive-lottie
https://github.com/zlgopen/awtk-android

xianjimli, 2021-06-15:
Ok, thank you. It’s too busy lately, I will look at it this weekend.

xianjimli, 2021-07-04:
AWTK rive player : https://github.com/zlgopen/awtk-widget-rive

ajsb85, 2021-07-04:
Good job @xianjimli 
Could you add the license headers in the repository?
I shared the repository with the community on Twitter
https://twitter.com/ajsb85/status/1411794785006989315

xianjimli, 2021-07-05:
Thank you. license is added.

closed 2021-07-05 by ajsb85