zlgopen/awtk #564: 输入法候选字词控件`candidates`没有响应有前缀`hard_key:`的按键

ufbycd, 2021-06-17:
我想在软件盘上添加两个按键来实现`左右键切换不同的候选字`：
```xml
        <candidates button_style="candidates" x="0" y="0" w="100%" h="50" grab_keys="true" />
```
```xml
        <button repeat="300" name="hard_key:left" style="highlight" w="15%" text="<-"/>
        <button repeat="300" name="hard_key:right" style="highlight" w="15%" text="->"/>
```
但经测试，`candidates`并没有响应；而在x86_64平台上按物理键盘的左右方向键却是有响应的。

xianjimli, 2021-06-17:
对hard_key处理确实有问题，已经改了。但是只能发给编辑，不能发给候选字，即使grab_key=true，也只能发给当前按钮，没有意义。

你的需求具体是什么？我看看能不能处理。

ufbycd, 2021-06-18:
大概是想符合PC上的输入习惯：
- 输入的时候，空格键后第一个候选字词上屏（输入到文本框）；
- 按左右方向键来选择候选字词，按空格键后，选中的候选字词上屏；

xianjimli, 2021-06-20:
硬键盘上有方向键吗？用硬键盘上方向键更方便一点吧。

ufbycd, 2021-06-20:
没有实体键盘，而且屏幕小，候选字词不好直接选中

ufbycd, 2021-06-22:
我按这个需求，自己修改了下：

```xml
      <group_box children_layout="default(r=1,c=0,s=2,m=2)">
        <button name="page:symnum" style="highlight" w="15%" text="123"/>
        <button name="page:lower" style="highlight" w="10%"><image draw_type="icon" image="en" x="center" y="middle" w="100%" h="100%"/></button>
        <button repeat="300" name="space" w="55%" text="Space"/>
        <button name="action" style="highlight" w="20%" text="Return"/>
      </group_box>
```
```diff
From 8694be50f88ca044f2338ac7fbd743646e6ed8ce Mon Sep 17 00:00:00 2001
From: ChenShiSheng <chenshisheng@reliachina.com>
Date: Mon, 21 Jun 2021 17:59:20 +0800
Subject: [PATCH] fixed "hard_Key" can't be used to select candidate words;

the first candidate is selected by default;
---
 src/base/input_engine.c               |  1 +
 src/ext_widgets/keyboard/candidates.c | 61 ++++++++++++++++++---------
 src/ext_widgets/keyboard/candidates.h |  2 +-
 3 files changed, 43 insertions(+), 21 deletions(-)

diff --git a/src/base/input_engine.c b/src/base/input_engine.c
index 224926cc6..281a33ffa 100644
--- a/src/base/input_engine.c
+++ b/src/base/input_engine.c
@@ -161,6 +161,7 @@ ret_t input_engine_input(input_engine_t* engine, int key) {
       engine->keys.str[engine->keys.size] = '\0';
     } else {
       input_engine_reset_input(engine);
+      input_engine_dispatch_candidates(engine, 0);
       return RET_FAIL;
     }
   } else {
diff --git a/src/ext_widgets/keyboard/candidates.c b/src/ext_widgets/keyboard/candidates.c
index 7a8489427..26c165c2a 100644
--- a/src/ext_widgets/keyboard/candidates.c
+++ b/src/ext_widgets/keyboard/candidates.c
@@ -65,6 +65,7 @@ static ret_t candidates_on_button_click(void* ctx, event_t* e) {
           input_method_dispatch_candidates(im, suggest_words->words, suggest_words->words_nr, -1);
           if (suggest_words->words_nr > 0) {
             widget_set_focused(widget_get_child(button->parent, 0), TRUE);
+            candidates->selected = 0;
             candidates->is_suggest = TRUE;
           }
           log_debug("suggest_words->words:%s\n", suggest_words->words);
@@ -182,6 +183,12 @@ static ret_t candidates_update_candidates(widget_t* widget, const char* strs, ui
   return_value_if_fail(candidates != NULL && strs != NULL, RET_BAD_PARAMS);
   return_value_if_fail(candidates_ensure_children(widget, nr + 1) == RET_OK, RET_OOM);
 
+  // 默认选中第一个候选
+  if (nr > 0 && selected == 0 && ! candidates->is_suggest) {
+      selected = 1;
+  }
+
+  candidates->selected = selected;
   children = (widget_t**)(widget->children->elms);
 
   for (i = 0; i < nr; i++) {
@@ -311,29 +318,43 @@ static ret_t candidates_move_focus(widget_t* widget, bool_t next) {
   return_value_if_fail(candidates != NULL, RET_BAD_PARAMS);
   nr = candidates->candidates_nr;
 
-  WIDGET_FOR_EACH_CHILD_BEGIN(widget, iter, i)
-  if (iter->focused) {
-    if (next) {
-      next_focus = i + 1;
-    } else {
-      if (i > 0) {
[... 78 lines cut ...]
 
   candidates->select_by_num = TRUE;
+  candidates->selected = 0;
   candidates->hscrollable = hscrollable_create(widget);
   hscrollable_set_always_scrollable(candidates->hscrollable, TRUE);
 
diff --git a/src/ext_widgets/keyboard/candidates.h b/src/ext_widgets/keyboard/candidates.h
index 149077d72..eebd0303f 100644
--- a/src/ext_widgets/keyboard/candidates.h
+++ b/src/ext_widgets/keyboard/candidates.h
@@ -98,7 +98,7 @@ typedef struct _candidates_t {
   uint32_t event_id;
   hscrollable_t* hscrollable;
   uint32_t candidates_nr;
-
+  int32_t selected;
   bool_t is_suggest;
 } candidates_t;
 
-- 
2.32.0

[0001-fixed-hard_Key-can-t-be-used-to-select-candidate-wor.patch.txt](https://github.com/zlgopen/awtk/files/6690354/0001-fixed-hard_Key-can-t-be-used-to-select-candidate-wor.patch.txt)

```