zlgopen/awtk #566: Edit控件不能弹出键盘

wodingdong, 2021-06-26:
项目开发到一半，需要用到edit控件，

wodingdong, 2021-06-26:
最后我的解决办法是，再先建一个项目，首先久开始创建键盘和edit控件，这样能够编译通过，自动弹出键盘。
然后再把原来的项目的文件再复制过来，其他的没有修改，这样才可以自动弹出键盘。

xianjimli, 2021-06-27:
在PC上可以重现吗，方便的话把代码发给我看看。

wodingdong, 2021-06-28:
原项目的代码就不方便发给你了，但是我建了一个demo，也是出现这样的情况。我最开始用这个awtk的时候，看到这个软键盘这些的时候写了demo发现了这个问题，我以为是我少了什么资源造成的，现在要频繁用到键盘，在项目中期发现总是无法弹起键盘。
  我怎么发给你整个项目代码demo， 发邮箱吗？请提供地址可以吗？

xianjimli, 2021-06-28:
你在微信群或QQ里吗？

wodingdong, 2021-06-30:
我在微信群发了这个问题，也没人解决，就提了issue, 
我再发下代码到微信群里

xianjimli, 2021-06-30:
你加我微信，把demo发给我吧

soyzhc, 2021-06-30:
APP_TYPE为 APP_DESKTOP时，edit控件不会弹出键盘，你的demo可能是这种类型
修改为 APP_SIMULATOR类型即可，具体可参考awtk自带的demoui配置

wodingdong, 2021-06-30:
> APP_TYPE为 APP_DESKTOP时，edit控件不会弹出键盘，你的demo可能是这种类型
> 修改为 APP_SIMULATOR类型即可，具体可参考awtk自带的demoui配置

那APP_TYPE为 APP_DESKTOP就不能弹出键盘了？
你这个说法不一定是正确的
不过我在demo中修改你说的这样确实好像是对的，不显示定义APP_TYPE，不弹出，定义为 APP_DESKTOP不弹出，定义APP_SIMULATOR可以弹出，
不过在我的项目中，现在也没有定义APP_TYPE，也可以弹出