zlgopen/awtk #568: Failed to get device IDError code: 7

amiltonrsilva, 2021-06-27:
Hi. 

   How to fix that error code 7.

[image]

Ubuntu 16.04.7 LTS
Release:	16.04

Thank you in advance.

soyzhc, 2021-06-28:
Are you using VMware virtual machine？
AWTK Designer Ubuntu only support the IDE hard disk type，you can change the vm setting

amiltonrsilva, 2021-06-28:
No. It's not in virtual machine, It is installed directly in a hard drive partition. 
This error occurs just after the application ask for authentication.

[image]

xuchaoze, 2021-06-28:
Can you try this command in Terminal, and tell me the result?
lsblk --nodeps -no serial

amiltonrsilva, 2021-06-28:
lsblk --nodeps -no serial

191925446914

WD-WX41A59KEHTV

soyzhc, 2021-07-01:
It may be a bug reading hard disk ID, please wait for the next upgrade version.
Please pay attention to awtk.zlg.cn