zlgopen/awtk #571: 使用check_button, AWTK Designer 设计和编译出来的差距太大

wodingdong, 2021-06-30:
[image]

```
<check_button name="check_one"  h="100%" w="100%" x="0" y="0" auto_adjust_size="true" style:normal:icon_at="right">
          <label name="label" x="0" y="0" w="85%" h="100%" auto_adjust_size="true" word_wrap="true" line_wrap="true" style:normal:text_align_h="left" tr_text="sssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssssss"/>
        </check_button>
```
AWTK Designer 设计和编译出来的差距太大了

soyzhc, 2021-07-01:
我们先测试一下看看

AWTK Designer 可以使用预览功能，预览时可以选择渲染方式比如，OpenGLES，AGGE
你可以根据实际工程的运行平台选择预览

xuchaoze, 2021-07-01:
因为AWTK Designer设计时会忽略auto_adjust_size属性，也就是该属性在设计时无效。