zlgopen/awtk #575: HJR TK499

ajsb85, 2021-07-09:
Dear @xianjimli 

I bought this development board.
I will use for a IoT project.
Do you think this board is a good candidate for AWTK?

[image]

Development files:
https://whycan.com/t_5861.html

ajsb85, 2021-07-09:
I bought all these components:
https://es.aliexpress.com/item/4001095715528.html?spm=a2g0s.9042311.0.0.274263c0kMKnPX

xianjimli, 2021-07-10:
Yes, it is. The author has ported AWTK (reference: https://www.amobbs.com/thread-5720658-1-1.html), but I don't know if this board supports AWTK, but the porting is not difficult.

ajsb85, 2021-07-10:
Do you agrees if we sponsor this board to the AWTK?
Can you provide a shipping address?

xianjimli, 2021-07-11:
Thank you. There is no need. In China, it is very convenient to buy various development boards. If you have problems with porting, just contact me.