zlgopen/awtk #577: 关于AWTK中颜色的问题

yuanyuanyuanquan, 2021-07-12:
在AWTK中是否能向其他前端框架一样实现渐变色的设计？若能，实现方式是怎样的？

xianjimli, 2021-07-12:
目前还没支持，近期可能会支持。

xianjimli, 2021-09-24:
最新版本提供了有限支持：https://github.com/zlgopen/awtk/blob/master/docs/gradient.md

yuanyuanyuanquan, 2021-11-15:
关于控件hscroll_label的一点建议，建议将控件的滑动速率设置为可调，固定时间导致界面中若存在多个该控件，就有可能由于超出文本不一致，导致画面不协调的情况。下面是我做的一点调整：
diff --git a/src/ext_widgets/scroll_label/hscroll_label.c b/src/ext_widgets/scroll_label/hscroll_label.c
index 541bc674f..20b750858 100644
--- a/src/ext_widgets/scroll_label/hscroll_label.c
+++ b/src/ext_widgets/scroll_label/hscroll_label.c
@@ -247,13 +247,14 @@ ret_t hscroll_label_step(widget_t* widget) {
 
   /* 更改滑动框hscroll_label的滑动策略，根据超出文本长度（range）确定滑动时间 */
   /* 例如此时为 30ms 位移一个像素 */
-  hscroll_label->duration = range * 30;
+  // hscroll_label->duration = range * 30;
+  hscroll_label->speed = hscroll_label->duration * range;
 
-  if (hscroll_label->elapsed >= hscroll_label->duration) {
-    hscroll_label->elapsed = hscroll_label->duration;
+  if (hscroll_label->elapsed >= hscroll_label->speed) {
+    hscroll_label->elapsed = hscroll_label->speed;
   }
 
-  percent = (float_t)hscroll_label->elapsed / (float_t)(hscroll_label->duration);
+  percent = (float_t)hscroll_label->elapsed / (float_t)(hscroll_label->speed);
   if (hscroll_label->reversed) {
     percent = 1 - percent;
   }
@@ -316,7 +317,7 @@ static ret_t hscroll_label_on_timer(const timer_info_t* info) {
 
   ret = hscroll_label_step(widget);
 
-  if (hscroll_label->elapsed >= hscroll_label->duration) {
+  if (hscroll_label->elapsed >= hscroll_label->speed) {
     uint32_t lull = hscroll_label->lull;
 
     hscroll_label->elapsed = 0;
@@ -478,7 +479,8 @@ widget_t* hscroll_label_create(widget_t* parent, xy_t x, xy_t y, wh_t w, wh_t h)
 
   hscroll_label->xoffset = 0;
   hscroll_label->lull = 3000;
-  hscroll_label->duration = 0;
+  hscroll_label->duration = 30;
+  hscroll_label->speed = 0;
 
   hscroll_label->loop = FALSE;
   hscroll_label->yoyo = FALSE;

在hscroll_label.h的结构体中添加`int32_t speed`。

xianjimli, 2021-11-23:
已合并，谢谢支持。