zlgopen/awtk #578: 窗口透明的问题

duanchangstar, 2021-07-13:
此款芯片的显示有多个图层，一个是video层在最底层，一个是fb层，在最高层。
awtk在fb层，做一个ui控制。
video层就播放视频的。
只要吧awtk窗体和控件改成透明的，就可以看到video层的视频。我把它放到一个view的容器里。然后这个容器设置成透明的。
但是awtk打开其他view控件，然后再回到视频播放这个view控件，发现之前的view的控件残影，留在了透明的视频播放这个view容器上，导致无法显示video层的视频。

请问这个问题如何解决。
演示的效果就是这样的（视频可能还在审核）

https://www.bilibili.com/video/BV1RB4y1N71M/

WNsACE, 2021-07-13:
有定义 WITH_LCD_CLEAR_ALPHA  宏吗？
你有参考过 [how_to_use_mutable_image.md](https://github.com/zlgopen/awtk/blob/master/docs/how_to_use_mutable_image.md) 文档么？

duanchangstar, 2021-07-13:
加上WITH_LCD_CLEAR_ALPHA 就OK了。感谢

duanchangstar, 2021-07-13:
> 有定义 WITH_LCD_CLEAR_ALPHA 宏吗？
> 可以参考 [how_to_use_mutable_image.md](https://github.com/zlgopen/awtk/blob/master/docs/how_to_use_mutable_image.md) 文档

加上WITH_LCD_CLEAR_ALPHA 就OK了。感谢

xchehub, 2022-09-30:
我的问题与楼主类似
视频层是在 fb0, awtk 应用是在 fb1
将awtk窗体和控件改成透明的，就可以看到video层的视频

不过在鼠标滑过透明区域时会有残影

我将 awtk/src/base/awtk_config_sample.h 复制一份成 awtk_config.h
放在同样位置

启用这二项设定
#define ENABLE_CURSOR 1
#define WITH_LCD_CLEAR_ALPHA 1

不知这样启用的作法是正确的吗？

xchehub, 2022-10-03:
啊
原来是我搞错方向
WITH_LCD_CLEAR_ALPHA 是得在 awth_config.py 中 声明
而非是 awth_config.h

问题解决了
谢谢