zlgopen/awtk #581: Failed to compile AWTK on Windows 10

ajsb85, 2021-07-21:
[image]
[image]

WNsACE, 2021-07-22:
这个应该是 vs2019 的编译环境没有安装好，应该是缺了一些头文件或者类库导致的，可以去研究一下那些东西没有安装好

ajsb85, 2021-07-22:
Thanks about the feedback.
The dependency list for Windows 10 is not published.

WNsACE, 2021-07-22:
我也是用 win10 和 vs2019 的，我是正常可以编译运行 awtk 的，这是我的 vs2019 的配置
[image]
[image]