zlgopen/awtk #582: 请问支持黑白屏吗

zxc3489, 2021-07-23:
我在awtk_config_sample.h的示例里只找到了RGB565和RGBA8888相关宏定义，并没有类似于黑白点阵的宏定义，请问支持黑白屏吗

xianjimli, 2021-07-23:
可以看看这个文档：https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_mono_lcd.md

zxc3489, 2021-07-31:
谢谢。这个文档看过了，也实践过了。可以。请问下用fontgen裁减下来的ttf，里面的size和mvvm的xml里面的size有什么关系呢？比如我为用fontgen生成了size为10的字库，在xml里指定了font_size分别为8和16，最终是以字库的图像为标准，按照比例进行放大缩小吗?

zxc3489, 2021-07-31:
fontgen第二个参数是我自己整理的常用中英文字符集合，文本内容如下
 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_`abcdefghijklmnopqrstuvwxyz{|}~，。？！一乙二十丁厂七卜人入......
 按照../awtk/bin/fontgen design/default/fonts/default.ttf ~/桌面/chinesetable out.ttf 5 生成ttf文件替换原来的字库，在程序进行字符资源初始化时会崩溃呢。是我生成导入的文本文档的格式不对吗

xianjimli, 2021-07-31:
fontgen的用法，你参考：tools/font_gen/README.md

字体的大小是像素，XML里的字体大小也是像素。

zxc3489, 2021-08-01:
awtk-linux-fb不能开LCD='SDL_FB_MONO'吗，开了有编译报错
In file included from build/var/awtk/src/lcd/lcd_sdl2_mono.c:24:0:
/root/rk3308_linux_release/awtk/awtk/src/lcd/lcd_sdl2_mono.h:25:17: fatal error: SDL.h: No such file or directory
 #include <SDL.h>

zxc3489, 2021-08-02:
上述问题导入SDL相关头文件解决了。感谢

zxc3489, 2021-08-02:
想问下lcd_mono_t:data的值和驱动里的frambuffer是怎么对应的呢，似乎比真正的fb要大得多呢

xianjimli, 2021-08-02:
你看看bitmap_mono_set_pixel，应该就会明白了。

zxc3489, 2021-08-02:
还是没看懂，能稍微讲解下吗，如何转化

zxc3489, 2021-08-02:
我使用bitmap_mono_dump打印了一下data的数据，很异常，大概是以下形状，呈现在lcd上的效果就是上面一部分全黑，下面全白(实际上ui的效果应该是白底里有黑字，黑底里有白字)，能提供下思路是大概是哪一步出问题了吗?目前为mono专门提供的修改就是在awtk-linux-fb/awtk_config.py里添加了      LCD='SDL_FB_MONO'，字体只用pyftsubset作了裁减，没有用fontgen;ui也为 mono做了适配。上述修改在pc上运行正常(pc修改的是awtk/awtk_config.py).请问有什么思路？目前我的猜测是字库问题，ui上的呈现应该是上面部分                                                                                                                                  
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
                                                                                                                                        
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
[... cut ...]
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

zxc3489, 2021-08-02:
我使用bitmap_mono_dump打印了一下data的数据，很异常，大概是以下形状，呈现在lcd上的效果就是上面一部分全黑，下面全白(实际上ui的效果应该是白底里有黑字，黑底里有白字)，能提供下思路是大概是哪一步出问题了吗?目前为mono专门提供的修改就是在awtk-linux-fb/awtk_config.py里添加了 LCD='SDL_FB_MONO'，字体只用pyftsubset作了裁减，没有用fontgen;ui也为 mono做了适配。上述修改在pc上运行正常(pc修改的是awtk/awtk_config.py).请问有什么思路？目前我的猜测是字库问题，ui上的呈现应该是上面部分
###

oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo 
###

zxc3489, 2021-08-02:
我使用bitmap_mono_dump打印了一下data的数据，很异常，大概是以下形状，呈现在lcd上的效果就是上面一部分全黑，下面全白(实际上ui的效果应该是白底里有黑字，黑底里有白字)，能提供下思路是大概是哪一步出问题了吗?目前为mono专门提供的修改就是在awtk-linux-fb/awtk_config.py里添加了 LCD='SDL_FB_MONO'，字体只用pyftsubset作了裁减，没有用fontgen;ui也为 mono做了适配。上述修改在pc上运行正常(pc修改的是awtk/awtk_config.py).请问有什么思路？目前我的猜测是字库问题
**

oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo
oooooooooooooo
**

zxc3489, 2021-08-02:
之前的全白全黑问题已经解决(板子上的字库文件丢失了)，但是加上字库文件后，显示就是乱码。。连字母都不能正常显示。。
我现在能做到lcd显示的内容与bitmap_mono_dump打印的图像保持一致。。。快cry了。

zxc3489, 2021-08-02:
mono的字库需要单独生成吗

xianjimli, 2021-08-02:
> mono的字库需要单独生成吗

是的。

zxc3489, 2021-08-03:
还是不行呢，我使用../../awtk/bin/fontgen default.ttf chinesetable default.data 14 mono生成了default.data文件，然后在application.c里引入资源文件#include "../source/default.data",在application_init()里添加assets_manager_add(assets_manager(), font_default);，还是字体还是不能正常显示呢

zxc3489, 2021-08-03:
请问字体加载和绘制可以看哪一部分代码呢

zxc3489, 2021-08-03:
搞定!在awtk-linux-fb的awtk_config.py中加入WITH_LCD_MONO宏解决，感谢解惑

closed 2021-08-03 by zxc3489