zlgopen/awtk #584: 使用cario时碰到的一些问题

hantianheng, 2021-08-06:
你好，最近看到awtk支持了rive动画，于是我们打算在自己的设备上面支持起来，因此在编译时矢量绘图库我们就使用了cario，但是发现使用cario时存在一些问题【以下问题均为使用cario时出现】。

**问题描述：**
切换主题后，rive动画显示不完整。下面分别是切换主题前后的显示效果。

[image]
[image]

另外切换主题后，会出现一些图标消失不见的问题。例如我们在设备的首页上面有个菜单按键，切换主题后菜单按键的图标没有显示出来，但是仍然可以点击。

菜单按键的实现方式：
` <button name="open:menu" style="blank_button" x="r" y="0" w="120" h="60" >//最外层是点击区域
      <button x="r:15" y="15" w="13" h="13" style="main_to_menu" /> //这里绘制一个圆形小图标
   </button>`

**复现方式**
将awtk-widget-rive/demos/window_main.c中的application_init()函数修改为：

`ret_t application_init(void) {
  rive_register();
  // bug when use cairo
  widget_set_theme((widget_t*)window_manager(),"default");

  widget_t* win = window_open("main");
  widget_child_on(win, "play", EVT_CLICK, on_play, win); 
  widget_child_on(win, "pause", EVT_CLICK, on_pause, win); 
  widget_child_on(win, "stop", EVT_CLICK, on_stop, win); 
  widget_child_on(win, "assets", EVT_VALUE_CHANGED, on_asset_changed, win); 

  window_manager_set_show_fps(window_manager(), TRUE);

  return RET_OK;
}`

**rive控件本身存在一个问题**

rive控件在销毁的时候没有关闭定时器，所以理论上存在系统死机的风险。
自己做了一些修改
awtk-widget-rive/src/rive/rive.cpp
`static ret_t rive_on_destroy(widget_t* widget) {
  rive_t* rive = RIVE(widget);
  return_value_if_fail(widget != NULL && rive != NULL, RET_BAD_PARAMS);
  RiveInfo* info = static_cast<RiveInfo*>(rive->info);

  delete info;
  rive_set_url(widget, NULL);

  if (rive->timer_id != TK_INVALID_ID) {
    timer_remove(rive->timer_id);
    rive->timer_id = TK_INVALID_ID;
  }
  return RET_OK;
}
`

xianjimli, 2021-08-06:
好的，谢谢。我看看。
widget_add_timer增加的timer，在widget destroy时会自动移除。

hantianheng, 2021-08-06:
好的

> 好的，谢谢。我看看。
> widget_add_timer增加的timer，在widget destroy时会自动移除。

好的

xianjimli, 2021-08-06:
设置主题的问题已经修复。麻烦有空更新代码试试，非常感谢。

hantianheng, 2021-08-08:
> 设置主题的问题已经修复。麻烦有空更新代码试试，非常感谢。

好的，谢谢