zlgopen/awtk #585: 关于edit的聚焦问题

zxc3489, 2021-08-15:
你好，我现在遇到一个比较头大的问题，在任一界面A进行edit的编辑的时候，如果这个时候弹出一个新界面B(业务需要，比如有呼叫呼入)，那么这个时候，焦点还是在界面A上，这个时候进行按键响应的还是界面A绑定的函数呢。该如何解决这个问题呢?edit(keyboard)的焦点控制在mvvm框架下是否能手动关闭呢?

xianjimli, 2021-08-16:
AWTK是最新版本吗？

zxc3489, 2021-08-16:
七月底的代码也试过了，一样的问题

xianjimli, 2021-08-16:
我加了一个测试，定时弹出一个对话框，没有发现问题：https://github.com/zlgopen/awtk-regression-test/blob/main/src/window_edit.c

你发一个可以重现的最小demo给我吧(在github上建一个参考也可以)。谢谢。

zxc3489, 2021-08-16:
我写demo的时候也没有复现，我去把keyboard的配置同步下试试，有新发现再联系您

zxc3489, 2021-08-16:
我基于awtk-mvvm-c-hello写了个demo，复现了问题:https://github.com/zxc3489/awtk-debug

xianjimli, 2021-08-16:
我没有重现，这样操作的：
* 1.在编辑器输入1
* 2.等待定时器打开新界面
* 3.在新界面用键盘输入123
* 4.退回之前的界面，之前界面仍然是1

zxc3489, 2021-08-16:
看下控制台的输出，聚焦在phone_id的时候，进入新界面，按"上"或者"下"按钮，控制台会有user_add_key_up/user_add_key_down函数里的log输出；聚焦在edit上时，进入新界面，同样操作，则没有相应的log输出

zxc3489, 2021-08-16:
在主界面时也要用上下键切换聚焦呢

zxc3489, 2021-08-16:
这边也给我提供下思路呢，我想找mvvm内部维护的widget(or page)的链表(从体验上感觉应该是维护了一个窗口的链表)，一直没找到门路呢。

xianjimli, 2021-08-17:
你绑定了焦点属性，数据是异步更新的，导致新窗口打开后，去设置旧窗口的焦点。之前AWTK处理有问题，你更新AWTK的代码试试。谢谢。

zxc3489, 2021-08-18:
更新到今天早上，一样的问题。不能绑定焦点属性?

xianjimli, 2021-08-18:
看看widget.c中，有没有下面的代码：
```c
 733 ret_t widget_set_focused_internal(widget_t* widget, bool_t focused) {
 734   widget_t* win = widget_get_window(widget);
 735   int32_t stage = widget_get_prop_int(win, WIDGET_PROP_STAGE, WINDOW_STAGE_NONE);
 736   return_value_if_fail(widget != NULL, RET_BAD_PARAMS);
 737 
 738   if (WINDOW_STAGE_SUSPEND == stage) {
 739     log_debug("You can not set focus of a widget when window is in background");
 740     return RET_FAIL;
 741   }
```

如果有，把awtk/awtk-mvvm/app clean一下，再重新编译试试。

closed 2021-10-19 by zxc3489