zlgopen/awtk #588: 屏幕旋转问题

Sokyx, 2021-08-20:
程序刚启动时，会使用tk_set_lcd_orientation旋转270，正常，在某个界面进入后需要对当前界面重新旋转，执行tk_set_lcd_orientation旋转非270的角度，0,90,180均会死机。打印image_copy的assert(dx >= 0 && (dx + src_r->w) <= dst->w);assert(dy >= 0 && (dy + src_r->h) <= dst->h);死机

Sokyx, 2021-08-20:
void dvp_gui_init(long icon_cache_size)
{
	printf("=====%s,%d=====\n", __FUNCTION__, __LINE__);
	
	tk_init(600, 1024, APP_MOBILE, NULL, "/data/awtkdemo/");
	tk_set_lcd_orientation(LCD_ORIENTATION_270);

	/* 初始化资源 */	
	assets_init();	
	/* 初始化扩展控件 */	
	tk_ext_widgets_init();
}

int main(int argc, char* argv[])
{
	int i = 0;
	
	dvp_gui_init(0);
	window_open("system_bar");
	widget_t* win = window_open("stair_main");

	tk_set_lcd_orientation(LCD_ORIENTATION_90);//注释掉就不会死机
	tk_run();
	return 0;
}

soyzhc, 2021-08-21:
在什么平台上使用, 用的是哪个版本的awtk?

Sokyx, 2021-08-21:
> 在什么平台上使用, 用的是哪个版本的awtk?

你好，是linux平台，awtk和awtk-linux-fb都是最新的，我一直都是更新的master分支，这个是更新到最新的master分支代码。更新之前是可以旋转的，更新之后第二次旋转就必死

WNsACE, 2021-08-23:
我今天测试了，应该以前的版本也不支持连续 lcd 旋转才对，你是用哪个版本测试可以的？可以提供一下可以连续 lcd 旋转的 awtk 版本给我测试么？

Sokyx, 2021-08-23:
> 我今天测试了，应该以前的版本也不支持连续 lcd 旋转才对，你是用哪个版本测试可以的？可以提供一下可以连续 lcd 旋转的 awtk 版本给我测试么？

我一直是下拉master分支的，具体不知道是哪个commit节点了，只知道是6个月之前，也就是二月份，那时候应该是发布了1.6版本，我这边上次更新记录的时候更新AWTK-1.6版本。如果一直都不支持多次旋转，但是我这边用的时候还可以旋转，奇怪。是不是当时我在rotate_image.inc文件里修改了一点：
case LCD_ORIENTATION_270: {
      for (i = 0; i < h; i++) {
        pixel_dst_t* s = src_p;
        pixel_dst_t* d = dst_p;

        for (k = 0; k < w; k++) {
			if((k<w-1) || i > 0)--------》这个是我加的，不然旋转270度会死机，当时我还提了一个issue旋转270死机
          *d = *s++;
          d = (pixel_dst_t*)(((char*)d) + dst_line_length);
        }

        dst_p--;
        src_p = (pixel_dst_t*)(((char*)src_p) + src_line_length);
      }
      break;
    }
但是这个旋转bug后面你们改了：
case LCD_ORIENTATION_270: {
      dx = src->h - src_r->y - 1;--------》这里是后面你们修复了
      dy = src_r->x;
      break;
    }

WNsACE, 2021-08-23:
最新的代码，已经增加了这个功能了，从代码的角度来说，先转 90 度，再转为 270 度的话，lcd 的宽高就会乱，以前的代码就有这个问题，所以本质来说不应该可以多次调用的

Sokyx, 2021-08-23:
嗯嗯，之前是启动旋转270，后面都是旋转180，再转回270，没有90

Sokyx, 2021-08-23:
先270再180，这个以前可以吗，如果都不行的话，那运行中就不能换方向了，但是之前我确实可以

Sokyx, 2021-08-24:
> 最新的代码，已经增加了这个功能了，从代码的角度来说，先转 90 度，再转为 270 度的话，lcd 的宽高就会乱，以前的代码就有这个问题，所以本质来说不应该可以多次调用的

8-23修改了lcd多次旋转后，不会死机，但是屏幕下方有一块区域刷新有问题

closed 2021-08-25 by Sokyx