zlgopen/awtk #596: scons  编译awtk失败   VS2019

ZhangYiXiSucceed, 2021-08-29:
scons  编译awtk失败   VS2019  ,  nodeJs  等组件都安装好 了

xianjimli, 2021-08-30:
具体什么问题？安装VS2019时，C/C++组件勾选了吗？

ZhangYiXiSucceed, 2021-08-30:
[image]
[image]

ZhangYiXiSucceed, 2021-08-30:
[image]

ZhangYiXiSucceed, 2021-08-30:
C/C++组件都是有的，编译到一半的时候 ，突然就出现这个问题了  说是debug 和 release 不匹配 就编译出错了

ZhangYiXiSucceed, 2021-08-30:
2019 和2015是不是有区别啊   那个md里面只是说安装>2015  感觉你们是2015编译的，所以还是有区别的

ZhangYiXiSucceed, 2021-08-30:
[image]

xianjimli, 2021-08-31:
我用的是2019。你可以改成不同的值试试。

ZhangYiXiSucceed, 2021-08-31:
这个每次编译时间都很长啊，感觉不是增量编译啊   有什么比较好的方法编译吗  还是就这一种

ufbycd, 2021-08-31:
AWTK的构建系统确实一般，我都提好多意见了。貌似李大侠相较于python更喜欢javascript，所以构建系统里有好多javascript写的工具，这就导致很多东西没有形成依赖关系。另外很多编译配置都是写死在构建脚本里面的，不够灵活。
像RT-Thread的构建系统就很完善：kconfig + scons，配置灵活、依赖清晰、功能完善。
另外一个问题是，AWTK不接受PR,有点闭门造车的感觉。

ZhangYiXiSucceed, 2021-08-31:
VS2019  需要更改 一会儿debug  一会儿release  就可以编成功了  这也很奇怪。。。。。

ZhangYiXiSucceed, 2021-08-31:
哥们，是的，很多都是脚本写的配置关系，我也没找到说明文档，然后碰到问题就得百度，