zlgopen/awtk #598: 关于slide_indicator_set_indicated_target 的使用

18611429192, 2021-09-02:
我想实现的东西是桌面有三个view 装在一个silde_view中，home_page在slide_view中第二页，但是我想让机器一开机就显示home_page而不是slide_view中的第一页，所以想用slide_indicator_set_indicated_target 这个函数在子控件初始化结束后将页面设置到home_page上，但是出现了问题，且不知道怎么结局，感觉官方文档对这些函数的描述太少了，摸不清楚怎么使用。
[image]
[image]
[image]

18611429192, 2021-09-02:
使用slide_view_set_active会直接崩溃
[image]

xianjimli, 2021-09-02:
你把窗口当slide_view使用了。先用widget_lookup查到slide_view，再调用slide_view_set_active。

18611429192, 2021-09-02:
明白了，感谢解答

closed 2021-09-02 by 18611429192