zlgopen/awtk #602: 怎么加快slide_view的切换速度呢？

18611429192, 2021-09-09:
我的xml文件现在是这个样子
```
<slide_view name="slideview" anim_hint="translate(duration=50, easing=sin_inout)" x="0" y="0" w="100%" h="100%" vertical="false" loop="TRUE">
    <view x="0" y="0" w="100%" h="100%">

    </view>

    <view x="0" y="0" w="100%" h="100%">

    </view>

    <view x="0" y="0" w="100%" h="100%">

    </view>

</slide>
```
感觉现在的这个duration=50，并没有工作，，我的三个view切换速度一直没有变化，想问下怎么才能加快这三个view的切换速度呢？

我也尝试了在初始化的时候调用widget_set_animator_time_scale 函数，也没有作用在这个切换动画上。

xianjimli, 2021-09-09:
不能修改。只能修改TK_ANIMATING_TIME宏调整控件默认动画时间。而且设置为50效果非常差，绘制的帧太少，界面跳跃。

18611429192, 2021-09-09:
好的，谢谢，改成了300，看起来感觉舒服了不少，，我在低成本嵌入式linux上使用awtk，编译选项中的`NANOVG_BACKEND`是不是设置成AGGE速度会快一点？
[image]

18611429192, 2021-09-09:
好的，谢谢，改成了300，舒服很多

closed 2021-09-09 by 18611429192