zlgopen/awtk #609: 应用执行exit(0)后，打印awtk死机

Sokyx, 2021-09-15:
应用主动执行exit(0)，然后打印awtk死机：
#0  0x000a2cd4 in rotate_bgra8888_image ( )
#1  0x0005f09c in lcd_mem_linux_wirte_buff ( )
#2  0x000926c2 in lcd_mem_flush ( )

不知道你们有没有遇到过啊

xianjimli, 2021-09-15:
有什么规律吗

Sokyx, 2021-09-15:
> 有什么规律吗

就是每次应用主动执行exit(0)退出的时候，基本都会打印这个死机，但是我如果是执行tk_quit跳出main里的tk_run循环，这种方式退出应用就基本不会死机。不知道这个是我的个例还是都会有这种情况

xianjimli, 2021-09-16:
"主动执行exit(0)"应该不是你期望的吧。我的意思时什么情况下会"主动执行exit(0)"，是必然的，是随机的，还是在某些情况下才出现的。

Sokyx, 2021-09-16:
> "主动执行exit(0)"应该不是你期望的吧。我的意思时什么情况下会"主动执行exit(0)"，是必然的，是随机的，还是在某些情况下才出现的。

是我主动执行的，就是应用要升级的时候，收到升级指令后，我就主动exit(0)退出应用，然后升级程序就开始升级。后来换成收到升级程序后应用执行tk_quit，就不死机

xianjimli, 2021-09-16:
exit是在后台线程调用的吗，那可能是有问题的。最好调用tk_quit让APP正常退出。

Sokyx, 2021-09-16:
> exit是在后台线程调用的吗，那可能是有问题的。最好调用tk_quit让APP正常退出。

是的，是升级业务的线程调用的，目前已经改成了tk_quit退出了。不过之前没出现过这个问题啊，之前也没注意过这个问题。

closed 2021-09-16 by Sokyx