zlgopen/awtk #610: 怎么限制最高显示帧率？

18611429192, 2021-09-16:
我通过宏定义`__FB_ASYNC_SWAP`，开起了一个专门的线程交换buffer，代码只要有绘制好的界面就直接显示了，目前最高的帧率是60，但是一旦滑动起来帧率就掉到了40多，且一直在抖动，这样的60帧只能是在设备静止时出现，还不如直接限制帧率到40，这样还能节省一些操作。但是不知道怎么设置，或者给个方向，我去实现一下。
```
static void* fbswap_thread(void* ctx) {
  fb_info_t* fb = &s_fb;
  struct fb_var_screeninfo vi = (fb->var);

  log_info("display_thread start\n");

  while (!s_app_quited) {
    tk_semaphore_wait(s_sem_ready, -1);//获取绘制好的信号
    if (s_app_quited) {
      break;
    }

    tk_mutex_lock(s_lck_fblist);
    fb_taged_t* ready_fb = get_ready_fb();
    assert(ready_fb);
    int ready_fbid = ready_fb->fbid;
    tk_mutex_unlock(s_lck_fblist);

    vi.yoffset = ready_fbid * fb_height(fb);
    ioctl(fb->fd, FBIOPAN_DISPLAY, &vi);

#if __FB_WAIT_VSYNC
    int dummy = 0;
    ioctl(fb->fd, FBIO_WAITFORVSYNC, &dummy);
#endif

    tk_mutex_lock(s_lck_fblist);
    fb_taged_t* last_busy_fb = get_busy_fb();
    if (last_busy_fb) {
      last_busy_fb->tags = FB_TAG_SPARE;
      tk_semaphore_post(s_sem_spare);
    }
    ready_fb->tags = FB_TAG_BUSY;
    tk_mutex_unlock(s_lck_fblist);
  }
```

xianjimli, 2021-09-16:
```
/**
 * @method window_manager_set_max_fps
 * 限制最大帧率。
 *
 *> TK\_MAX\_LOOP\_FPS/max\_fps最好是整数，比如TK\_MAX\_LOOP\_FPS为120，max\_fps可取60/30/20/10等。
 *
 * @annotation ["scriptable"]
 * @param {widget_t*} widget 窗口管理器对象。
 * @param {uint32_t}  max_fps 最大帧率。
 *
 * @return {ret_t} 返回RET_OK表示成功，否则表示失败。
 */
ret_t window_manager_set_max_fps(widget_t* widget, uint32_t max_fps);
```

18611429192, 2021-09-16:
ok
谢谢

closed 2021-09-16 by 18611429192