zlgopen/awtk #611: 更换主界面背景图片的方法

18611429192, 2021-09-17:
我现在加了一个按钮在主界面，想实现的效果是: 按一下按钮主界面的背景图片就切换一次。
我找了背景图片的绘制函数，`widget_draw_background (widget_t* widget, canvas_t* c);`， 但是需要一个canvas_t指针，看到了`canvas_draw_image_at `函数，使用`widget_load_image(widget, "myimage", &bitmap);`加载图片到bitmap，然后绘制到canvas对象里，在绘制到背景，但是这么做，直接导致了程序崩溃。
后来我的做法是将不同的背景图片设置成了window的不同style，通过使用`widget_use_style`来设置不同的style，切换背景。但是这么做style更换了，但是不会自动更新，需要我关闭，再打开窗口，才能切换成功。这样的话感觉不是很对，有没有好的方法，更换窗口的背景图片呢？

xianjimli, 2021-09-17:
用widget_set_style_str设置背景就可以：

```
/**
 * @method widget_set_style_str
 * 设置字符串类型的style。
 *
 * > * [state 的取值](https://github.com/zlgopen/awtk/blob/master/docs/manual/widget_state_t.md)
 * > * [name 的取值](https://github.com/zlgopen/awtk/blob/master/docs/theme.md)
 *
 * @annotation ["scriptable"]
 * @param {widget_t*} widget 控件对象。
 * @param {const char*} state_and_name 状态和名字，用英文的冒号分隔。
 * @param {const char*} value 值。
 *
 * @return {ret_t} 返回RET_OK表示成功，否则表示失败。
 */
ret_t widget_set_style_str(widget_t* widget, const char* state_and_name, const char* value);
```

18611429192, 2021-09-17:
感谢，我去试试

closed 2021-09-17 by 18611429192