zlgopen/awtk #619: dialog弹窗设置single_instance="true"，后者覆盖前者，但是前者没有退出

Sokyx, 2021-10-09:
ret_t evawtk_message_box(const char* title, const char* text, int btn_count, int timeout)        
{
       widget_t* dialog = NULL;
	int ret = 0;
	int return_id = 0;

	dialog = dialog_open("comm_msg_box");

	_msgbox_init_priv_ctx();

	_msgbox_button_text_init(dialog, title, text, btn_count);

	_msgbox_widget_init(dialog);

#ifdef KEY_STAIR
	widget_on(dialog, EVT_KEY_UP, dialog_key_event, dialog);
#endif

	if((btn_count <= 0) && (timeout > 0))
    {
		widget_add_timer(dialog, timer_func, timeout);
		widget_on(dialog, EVT_POINTER_DOWN, dialog_clicked, dialog);
    }
	widget_on(dialog, EVT_DESTROY, dialog_quit_win, dialog);

	ret = dialog_modal(dialog);
	if(ret == DIALOG_QUIT_YES)
		return_id = RET_OK;
	else
		return_id = RET_FAIL;
	log_Debug("MSG_BOX", "==================>%s, title = %s, text = %s exit!\n", __func__, title, text);
        return return_id;
}

弹出的dialog是没有按钮的，弹窗的时候会执行widget_add_timer，添加一个2s的定时器，定时器里会执行dialog_quit(dialog, DIALOG_QUIT_YES);退出dialog
出现的情况是：
1、弹出一个dialog，2s之前弹出另一个dialog，只打印了一次退出，也就是第二个dialog的退出，第一个dialog并没有打印退出，而且也没有阻塞，是什么情况呢

Sokyx, 2021-10-09:
如果不设置single_instance="true"，那第二个弹窗退出了，第一个弹窗就一直都还在

xianjimli, 2021-10-09:
single_instance不支持模态对话框，因为模态对话框会阻塞消息循环。你要实现什么效果

xianjimli, 2021-10-09:
不要用模态

Sokyx, 2021-10-09:
> 不要用模态

不用模态的话，有些情况下打开一个界面，window_open("page1");  然后判断条件错误后弹出非模态对话框，由于不阻塞，瞬间就执行window_close关闭page1，速度太快会导致page1页面无法关闭。目前我是把dialog设置成同时只能打开一个，前者打开，后者直接return不执行dialog_open了