zlgopen/awtk #643: 透明窗体

nihjin, 2021-12-09:
第一个问题;先打开一个不透明窗口，然后再打开另外一个透明窗口并把之前不透明窗口关闭，显示是不透明窗口图片（应该是透明窗口，看不之前打开窗体才对）
第二个问题：怎么在UI上面显示视频(有视频层和UI层)

这两个问题怎么解决
谢谢

xianjimli, 2021-12-10:
参考一下这个文档：https://github.com/zlgopen/awtk/blob/master/docs/how_to_use_mutable_image.md

nihjin, 2021-12-14:
用户也可以在该函数中绘画其他的图像数据，就可以达到动态播放视频或者摄像头的效果。这个是把视频数据刷到这个UI上面。有没有办法，可以把UI层某区域透明，然后可以看到视频层数据(设备有两层，视频层和UI层)？

nihjin, 2021-12-14:
另外还有个问题有时候调用window_close(wid);
会报这个错误window_manager_default_close_window:537 condition(window_is_opened(window)) failed!关闭不了窗口

再过几分钟调用window_close(wid);又可以关掉窗口。是什么原因？

xianjimli, 2021-12-15:
> 有没有办法，可以把UI层某区域透明，然后可以看到视频层数据(设备有两层，视频层和UI层)？

把上面的文档看完。

xianjimli, 2021-12-15:
> 另外还有个问题有时候调用window_close(wid); 会报这个错误window_manager_default_close_window:537 condition(window_is_opened(window)) failed!关闭不了窗口
> 
> 再过几分钟调用window_close(wid);又可以关掉窗口。是什么原因？

警告显示的窗口没有打开，所以无法关闭。如果确定是AWTK处理流程有问题，弄个最简的demo给我看看。

Alvin-xie, 2022-02-25:
请问一下这个问题搞定了吗？ 我有同样的疑问，在非透明窗口1上，切换到透明窗口2上，窗口2透明部分显示的是窗口1的内容。但是如果增加一个过渡动画在窗口2就看不到窗口1的图片。
我这边按照 https://github.com/zlgopen/awtk-c-demos/blob/master/demos/mutable_image.c 这个demo做测试填充图片设置属性的时候 alpha通道的值没有作用。
  decoding->color = color_init(0xff, 0, 0, 0); ——显示的是白色图片
  decoding->image = bitmap_create_ex(w, h, 0, format);

WeSrc-Yin, 2022-03-22:
> 
我也遇到同样的问题。概率性的，快速切换几个窗口就比较容易出现。@xianjimli

WeSrc-Yin, 2022-03-22:
> 另外还有个问题有时候调用window_close(wid); 会报这个错误window_manager_default_close_window:537 condition(window_is_opened(window)) failed!关闭不了窗口
> 
> 再过几分钟调用window_close(wid);又可以关掉窗口。是什么原因？
我也遇到同样的问题。概率性的，快速切换几个窗口就比较容易出现。
@xianjimli

xianjimli, 2022-03-30:
> > 另外还有个问题有时候调用window_close(wid); 会报这个错误window_manager_default_close_window:537 condition(window_is_opened(window)) failed!关闭不了窗口
> > 再过几分钟调用window_close(wid);又可以关掉窗口。是什么原因？
> > 我也遇到同样的问题。概率性的，快速切换几个窗口就比较容易出现。
> > @xianjimli

窗口是异步打开的，可能还没有完全打开就去关闭它。