zlgopen/awtk #644: 移植到嵌入式stm32f103后显示异常

dengtong2018, 2021-12-13:
[image]
移植到stm32f103后显示这样是什么问题？

dengtong2018, 2021-12-13:
经测试把FRAGMENT_FRAME_BUFFER_SIZE这个buffer大小改成512字节后正常，但是刷频速度明显慢了很多，之前是8K

xianjimli, 2021-12-13:
可能是LCD宽度不对。

xianjimli, 2021-12-13:
> 经测试把FRAGMENT_FRAME_BUFFER_SIZE这个buffer大小改成512字节后正常，但是刷屏速度明显慢了很多，之前是8K

正常情况。buffer太小，就要画很多遍。

dengtong2018, 2021-12-13:
> 

为什么buffer改小后画面正常了

dengtong2018, 2021-12-13:
> 可能是LCD宽度不对。

改了LCD宽度会有白边，可能和这个没关系。

dengtong2018, 2021-12-13:
> > 经测试把FRAGMENT_FRAME_BUFFER_SIZE这个buffer大小改成512字节后正常，但是刷屏速度明显慢了很多，之前是8K
> 
> 正常情况。buffer太小，就要画很多遍。
为什么buffer改小后画面正常了

xianjimli, 2021-12-15:
可能是内存不够，导致内存分配失败。

Ilikecoppy, 2022-03-08:
我遇到了同样的问题，发现是widget_init里面的widget->dirty_rect_tolerance = 4;这里的设置会导致计算脏矩形的时候算出一些奇奇怪怪的值，我改成8就没出现问题了