zlgopen/awtk #649: deepin os 段错误

mankouzhuya, 2021-12-15:
操作系统:
[image]
master分支编译后运行demo，报段错误
(demoui:38190): GLib-GObject-CRITICAL **: 20:48:39.686: g_value_set_boxed: assertion 'G_VALUE_HOLDS_BOXED (value)' failed

(demoui:38190): dbind-WARNING **: 20:48:39.702: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
段错误

[image]

mankouzhuya, 2021-12-16:
修改awt_config.py之后重新编译，还是报段错误
[image]

xianjimli, 2021-12-18:
可能是opengl驱动不兼容。修改awtk_config.py, 换成cairo试试：
```
 32 # NANOVG_BACKEND='GLES2'
 33 # NANOVG_BACKEND='GLES3'
 34 # NANOVG_BACKEND='AGG'
 35 # NANOVG_BACKEND='BGFX'
 36 # NANOVG_BACKEND='AGGE'
 37 VGCANVAS='CAIRO'
 ```

mankouzhuya, 2021-12-21:
切换为Ubuntu系统好了。应该是deepin系统有问题
[image]

xianjimli, 2021-12-22:
好的。