zlgopen/awtk #651: 在win10上打开编译好的应用，输入法就会切换到中文模式

Arkinux, 2021-12-22:
我使用AWTK开发windows桌面应用程序，我的win10上使用微软拼音输入法，默认是英文模式，但是打开编译好的AWTK桌面应用程序后，win10的输入法就会自动切换到中文模式，请问如何在打开AWTK开发的桌面应用程序时不切换win10的输入法模式？

xianjimli, 2021-12-23:
输入法是调用SDL的接口，貌似它没有提供切换输入模式函数。

Arkinux, 2021-12-24:
噢，那有没有什么好办法在打开AWTK应用时不自动切换输入模式呢？因为我做的是一个扫码器扫码条形码输入的应用，如何是在中文模式下扫描器扫出来的字符串就不对

xianjimli, 2021-12-27:
编辑器不得到焦点，应该不会打开输入法啊。

Arkinux, 2022-01-01:
我说的是windows的输入法，当前windows的输入法模式默认是英文模式，但打开AWTK应用后就会自动切换输入法模式为中文模式，这是我打开AWTK应用的录屏过程，有没有什么好办法在打开AWTK应用时不自动切换输入模式呢？[image]

RainMic1019, 2022-01-04:
> 我说的是windows的输入法，当前windows的输入法模式默认是英文模式，但打开AWTK应用后就会自动切换输入法模式为中文模式，这是我打开AWTK应用的录屏过程，有没有什么好办法在打开AWTK应用时不自动切换输入模式呢？[image]

这个可能是由于Windows 10系统中将默认输入法设置为“中文（微软拼音）”导致的，比如下图：

[image]

将其修改成“英语（美式键盘）”应该能解决这个问题。