zlgopen/awtk #655: Can't build pass with the latest code on Rasberry Pi

wnnwoo, 2022-01-01:
I suffer this build error when I build with the latest code.
It seems there is a method to add special pkg-conig in the build script.

In file included from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /usr/include/gtk-3.0/gdk/gdkconfig.h:13,
                 from /usr/include/gtk-3.0/gdk/gdk.h:30,
                 from /usr/include/gtk-3.0/gtk/gtk.h:30,
                 from 3rd/SDL/src/video/x11/SDL_x11window.c:36:
/usr/include/glib-2.0/glib/gtypes.h:32:10: fatal error: glibconfig.h: No such file or directory
   32 | #include <glibconfig.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
scons: *** [3rd/SDL/src/video/x11/SDL_x11window.o] Error 1
scons: building terminated because of errors.

WNsACE, 2022-01-02:
在树莓派上面没有试过使用 SDL 编译，而且 SDL 编译要依赖相对多的类库，上面的提示是依赖 x11 出错，你的树莓派有安装 x11 和 gdk-3 么？

soyzhc, 2022-01-02:
try this
```
sudo ln -s /usr/lib/arm-linux-gnueabihf/glib-2.0/include/glibconfig.h /usr/include/glib-2.0/
```