zlgopen/awtk #657: Mac M1版本电脑无法编译通过

calvingit, 2022-01-11:
```bash
ite_crt.o 3rd/mbedtls/library/x509write_csr.o 3rd/mbedtls/library/xtea.o
ranlib lib/libmbedtls.a
ar rc lib/libnanovg.a 3rd/nanovg/base/nanovg.o
ranlib lib/libnanovg.a
ar rc lib/libglad.a 3rd/glad/glad.o
ranlib lib/libglad.a
g++ -o bin/action_thread_test -framework IOKit -framework Cocoa -framework QuartzCore -framework OpenGL -weak_framework Metal -weak_framework MetalKit tests/action_thread_test.o -Llib -Lbin -L/usr/local/lib -lassets -limage_gen -ltheme_gen -lfont_gen -lstr_gen -lcommon -lawtk_global -lextwidgets -lwidgets -lbase -lgpinyin -lfribidi -llinebreak -lfscript_ext -lstreams -lconf_io -lhal -lxml -lcharset -lcsv -lubjson -lcompressors -lminiz -ltkc_core -lmbedtls -lnanovg -lSDL2 -lglad -lstdc++ -liconv -lpthread -lm -ldl
ld: library not found for -lSDL2
clang: error: linker command failed with exit code 1 (use -v to see invocation)
scons: *** [bin/action_thread_test] Error 1
scons: building terminated because of errors.
```

SDL2安装成功:
```bash
➜  brew list sdl2
/opt/homebrew/Cellar/sdl2/2.0.18/bin/sdl2-config
/opt/homebrew/Cellar/sdl2/2.0.18/include/SDL2/ (77 files)
/opt/homebrew/Cellar/sdl2/2.0.18/lib/libSDL2-2.0.0.dylib
/opt/homebrew/Cellar/sdl2/2.0.18/lib/cmake/ (2 files)
/opt/homebrew/Cellar/sdl2/2.0.18/lib/pkgconfig/sdl2.pc
/opt/homebrew/Cellar/sdl2/2.0.18/lib/ (4 other files)
/opt/homebrew/Cellar/sdl2/2.0.18/share/aclocal/sdl2.m4
```

xianjimli, 2022-01-12:
在macpro(2015)上，SDL在这个位置：
jim@lixianjings-MacBook-Pro awtk % otool -l bin/action_thread_test|grep SDL
         name /usr/local/opt/sdl2/lib/libSDL2-2.0.0.dylib (offset 24)

xianjimli, 2022-01-12:
你修改一下awtk_config_common.py，在91行加入/opt/homebrew/Cellar/sdl2/2.0.18/lib/试试：

```
OS_LIBPATH = ['/usr/local/lib/','/opt/homebrew/Cellar/sdl2/2.0.18/lib/'] 
```

calvingit, 2022-01-12:
OK，可以跑demo了

closed 2022-01-12 by calvingit