zlgopen/awtk #661: 问题咨询:label控件在中文内容过长时属性勾选自动换行导致崩溃（英文没有影响）

Sunyhs, 2022-01-25:
xml代码如下：

<popup x="c" y="m" w="35%" style:normal:round_radius="4" h="50%">
  <view name="view" x="8%" y="6%" w="90%" h="80%" children_layout="default(r=0,c=1,x=0,y=10,s=10)">
    <label name="label1" h="33%" line_wrap="true" word_wrap="true" style="default" text="中文测试短"/>
    <label name="label2" h="33%" line_wrap="true" word_wrap="true" style="default" text="中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中文测试长"/>
    <label name="label3" h="33%" line_wrap="true" word_wrap="true" style="default" text="showshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshow"/>
  </view>
</popup>

gdb调试结果如下：
(gdb) bt
#0  0x00241838 in label_on_paint_self ()
#1  0x0028bed8 in set_linebreaks.constprop ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb)

Sunyhs, 2022-01-25:
xml代码如下：
<popup x="c" y="m" w="35%" style:normal:round_radius="4" h="50%">
  <view name="view" x="8%" y="6%" w="90%" h="80%" children_layout="default(r=0,c=1,x=0,y=10,s=10)">
    <label name="label1" h="33%" line_wrap="true" word_wrap="true" style="default" text="中文测试短"/>
    <label name="label2" h="33%" line_wrap="true" word_wrap="true" style="default" text="中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中文测试长"/>
    <label name="label3" h="33%" line_wrap="true" word_wrap="true" style="default" text="showshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshow"/>
  </view>
</popup>

Sunyhs, 2022-01-25:
<popup x="c" y="m" w="35%" style:normal:round_radius="4" h="50%">
  <view name="view" x="8%" y="6%" w="90%" h="80%" children_layout="default(r=0,c=1,x=0,y=10,s=10)">
    <label name="label1" h="33%" line_wrap="true" word_wrap="true" style="default" text="中文测试短"/>
    <label name="label2" h="33%" line_wrap="true" word_wrap="true" style="default" text="中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中文测试长"/>
    <label name="label3" h="33%" line_wrap="true" word_wrap="true" style="default" text="showshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshow"/>
  </view>
</popup>

Sunyhs, 2022-01-25:
> xml代码如下：
<popup x="c" y="m" w="35%" style:normal:round_radius="4" h="50%">
  <view name="view" x="8%" y="6%" w="90%" h="80%" children_layout="default(r=0,c=1,x=0,y=10,s=10)">
    <label name="label1" h="33%" line_wrap="true" word_wrap="true" style="default" text="中文测试短"/>
    <label name="label2" h="33%" line_wrap="true" word_wrap="true" style="default" text="中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中文测试长"/>
    <label name="label3" h="33%" line_wrap="true" word_wrap="true" style="default" text="showshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshow"/>
  </view>
</popup>

Sunyhs, 2022-01-25:
`<popup x="c" y="m" w="35%" style:normal:round_radius="4" h="50%">
  <view name="view" x="8%" y="6%" w="90%" h="80%" children_layout="default(r=0,c=1,x=0,y=10,s=10)">
    <label name="label1" h="33%" line_wrap="true" word_wrap="true" style="default" text="中文测试短"/>
    <label name="label2" h="33%" line_wrap="true" word_wrap="true" style="default" text="中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中文测试长"/>
    <label name="label3" h="33%" line_wrap="true" word_wrap="true" style="default" text="showshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshow"/>
  </view>
</popup>`

Sunyhs, 2022-01-25:
`xml代码如下: （加了换行方便阅读）
<popup x="c" y="m" w="35%" style:normal:round_radius="4" h="50%">

  <view name="view" x="8%" y="6%" w="90%" h="80%" children_layout="default(r=0,c=1,x=0,y=10,s=10)">

    <label name="label1" h="33%" line_wrap="true" word_wrap="true" style="default" text="中文测试短"/>

    <label name="label2" h="33%" line_wrap="true" word_wrap="true" style="default" text="中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中文测试长"/>

    <label name="label3" h="33%" line_wrap="true" word_wrap="true" style="default" text="showshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshow"/>

  </view>

</popup>`

Sunyhs, 2022-01-25:
<label name="label1" h="33%" line_wrap="true" word_wrap="true" style="default" text="中文测试短"/>
<label name="label2" h="33%" line_wrap="true" word_wrap="true" style="default" text="中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中文测试长"/>
<label name="label3" h="33%" line_wrap="true" word_wrap="true" style="default" text="showshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshow"/>

Sunyhs, 2022-01-25:
`
 <label name="label1" h="33%" line_wrap="true" word_wrap="true" style="default" text="中文测试短"/>
 <label name="label2" h="33%" line_wrap="true" word_wrap="true" style="default" text="中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中中文测试长"/>
<label name="label3" h="33%" line_wrap="true" word_wrap="true" style="default" text="showshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshowshow"/>
`

Sunyhs, 2022-01-25:
view中三个label控件，中文短测试， 中文长测试，英文长测试，崩溃以后将中文长测试剪短测试后，英文长实测通过

WNsACE, 2022-01-25:
你可以提供完整的崩溃的 ui 文件？我测试了你上面的那一段 xml 代码，没有出现崩毁问题，我用的是 1/24 号的 awtk 代码测试的，要么你可以试一下把 awtk 的版本升级看看还没有？

Sunyhs, 2022-01-25:
好的，我是用的之前版本大概是2021/07/22左右的版本，label中文内容过长时会导致崩溃，用window_open开启一个xml弹窗，xml中由popup创建，结构如此：<popup> <view> <label/> <label/> <label/> </view></popup>  label控件中的内容为我展示的内容

Sunyhs, 2022-01-25:
好的，我是用的之前版本大概是2021/07/22左右的版本，label中文内容过长时会导致崩溃，用window_open开启一个xml弹窗，xml中由popup创建，结构如此：popup   view   label,label,label   /view   /popup     label控件中的内容为我展示的内容

WNsACE, 2022-01-26:
最新的代码是正常么？如果不正常的话，你提供一个完整的 xml 文件？我这边测试一下