zlgopen/awtk #664: awtk很多控件比如progress_circle效率低下的问题

smiletigerpp, 2022-01-30:
在awtk中优化的很不好,虽然可以用,但是效率非常低下,比如一个progress_circle控件,粗暴的强制刷新该控件的长宽占用的矩形框,而非实际使用到的矩形框,这个控件一般用来指示进度,那么实际利用率仅仅为一小段弧形,并且每次更新的时候更加是弧形里面的一小段,但是awtk中每一帧更新的为该控件的长款的矩形,非常影响效率,在没有用该控件之前我可以刷新率达到125帧/秒,用了一个400*400的progress_circle控件之后,刷新率降到了恐怖的30帧,,可以学习一下touchgfx的circle控件,源码touhgfx里面都有,他就是会去获取一个实际用到的圆弧上的一小段,同样用400x400的控件,如果我实际上使用到的圆弧宽度为20的话,awtk直接每一帧率刷新400x400的区域,在touchgfx里面每一帧刷新的区域只有20x20不到,性能差了好几百倍.所以请求awtk能否考虑一下性能上的优化,而非一味地增加功能,毕竟awtk面向的是跨平台,性能还是首要的,

我这里可以附上touchgfx的算法,可以在touchgfx的源码里面找到

/**
  ******************************************************************************
  * This file is part of the TouchGFX 4.16.1 distribution.
  *
  * <h2><center>&copy; Copyright (c) 2021 STMicroelectronics.
  * All rights reserved.</center></h2>
  *
  * This software component is licensed by ST under Ultimate Liberty license
  * SLA0044, the "License"; You may not use this file except in compliance with
  * the License. You may obtain a copy of the License at:
  *                             www.st.com/SLA0044
  *
  ******************************************************************************
  */

#include <touchgfx/widgets/canvas/Circle.hpp>

namespace touchgfx
{
Circle::Circle()
    : CanvasWidget(),
      circleCenterX(0), circleCenterY(0), circleRadius(0),
      circleArcAngleStart(CWRUtil::toQ5<int>(0)), circleArcAngleEnd(CWRUtil::toQ5<int>(360)),
      circleLineWidth(0), circleArcIncrement(5),
      circleCapArcIncrement(180)
{
    Drawable::setWidthHeight(0, 0);
}

void Circle::setPrecision(int precision)
{
    if (precision < 1)
    {
        precision = 1;
    }
    if (precision > 120)
    {
        precision = 120;
    }
    circleArcIncrement = precision;
}

int Circle::getPrecision() const
{
    return circleArcIncrement;
}

void Circle::setCapPrecision(int precision)
{
    if (precision < 1)
    {
        precision = 1;
    }
    if (precision > 180)
    {
        precision = 180;
    }
    circleCapArcIncrement = precision;
}

int Circle::getCapPrecision() const
{
    return circleCapArcIncrement;
}

bool Circle::drawCanvasWidget(const Rect& invalidatedArea) const
[... 463 lines cut ...]
        // start is higher than end
        if (endAngleQ5 < circleArcAngleEnd)
        {
            // end moved even lower
            minAngle = endAngleQ5;
            maxAngle = CWRUtil::Q5(ROUNDUP((int)circleArcAngleEnd, circleArcIncrementQ5int));
            maxAngle = MIN(maxAngle, circleArcAngleStart); // No need to go higher than start
        }
        else if (endAngleQ5 < circleArcAngleStart)
        {
            // end moved higher, but not higher than start
            minAngle = circleArcAngleEnd;
            maxAngle = CWRUtil::Q5(ROUNDUP((int)endAngleQ5, circleArcIncrementQ5int));
            maxAngle = MIN(maxAngle, circleArcAngleStart);
        }
        else
        {
            // end moved past start
            minAngle = circleArcAngleEnd;
            maxAngle = endAngleQ5;
        }
    }
    return getMinimalRect(minAngle, maxAngle);
}
} // namespace touchgfx

xianjimli, 2022-02-05:
你帮我用这个分支测试吧，谢谢：https://github.com/zlgopen/awtk/tree/opt_progress_circle

smiletigerpp, 2022-02-05:
好的 愿意帮忙测试 希望awtk越来越好 这个是不是还跟脏矩形那边lcd.inc那个也需要优化。这个刷新区域是脏矩形那边获取判断的吧

smiletigerpp, 2022-02-05:
> 你帮我用这个分支测试吧，谢谢：https://github.com/zlgopen/awtk/tree/opt_progress_circle

我测试了一下比之前好很多很多了,已经可以用在仪表上面60帧率了,但是还是觉得有优化的空间,比如400x400的控件,然后linewidth设置为100,如果我背景色为透明色,前景为图片,实际前景图片只有线宽10的有效区域,那么实际上percent的控件还是按照100线宽来计算脏矩形的,是否可以智能的去判断透明区域,把透明区域减掉,这样还可以继续优化比较大的空间,就是前景背景在处理透明的时候要把透明区域剪裁掉,当然我只是用printf来debug测试了一下,还没有用到jlink来具体测试,我会继续关注的,非常感谢大佬用心过年还在加班解决优化

xianjimli, 2022-02-06:
好的，谢谢。

“linewidth设置为100,如果我背景色为透明色,前景为图片,实际前景图片只有线宽10的有效区域”有实际需求，还是只是方便一点？“智能的去判断透明区域”也是需要花时间的，我也不想把代码弄得太复杂。

smiletigerpp, 2022-02-06:
> 好的，谢谢。
> 
> “linewidth设置为100,如果我背景色为透明色,前景为图片,实际前景图片只有线宽10的有效区域”有实际需求，还是只是方便一点？“智能的去判断透明区域”也是需要花时间的，我也不想把代码弄得太复杂。

实际上这个控件很重要的 经常用在仪表盘上面的 最最关键的一个控件 用来配合指针旋转背景用的 你们awtk那个配合rt1052还是imux6上面那个仪表demo没有用的这个功能 但是基本上所有的仪表控件都是需要这个功能的 这个控件拿来做百分比指示是大材小用了
[image]