zlgopen/awtk #669: 问题咨询：需要列出U盘中所有的音频文件，获取到上万个文件时，需要创建上万个button，那种控件可以动态加载这些button呢

liuning263, 2022-02-12:


xianjimli, 2022-02-13:
可以试试：https://github.com/zlgopen/awtk-widget-table-view

liuning263, 2022-02-14:
> 可以试试：https://github.com/zlgopen/awtk-widget-table-view

好的，非常感谢

liuning263, 2022-02-21:
使用table_view时，使用上下按键滚动table_row，当焦点超出当前可视row控件后会显示空白。

liuning263, 2022-02-22:
> 使用table_view时，使用上下按键滚动table_row，当焦点超出当前可视row控件后会显示空白。

这个问题解决了，由于使用的awtk代码比较旧，出现隐藏掉了后半部分table_row控件的问题，贴一下修改后的代码：在table_client.c中 
static ret_t table_client_prepare_data(widget_t* widget) {
  。。。。。。。。
  。。。。。。。。
  for (i = nr + start_index; i < max_rows_to_load; i++) {
    widget_t* iter = widget_get_child(widget, i);
    widget_set_visible(iter, FALSE, FALSE);
    widget_set_enable(iter, FALSE);
  }

  return RET_OK;
}     
修改了for循环中i的初始值。

liuning263, 2022-02-23:
> 可以试试：https://github.com/zlgopen/awtk-widget-table-view
目前还有一个问题，我的awtk版本比较旧了。打开界面后所有的table_row控件会显示在第一行，文字都会重叠在一起。滚动一下鼠标、滑动一下屏幕之后就好了。调用table_client_scroll_to（widget,0,duration）函数把   
if (yoffset_end == table_client->yoffset) {
    return RET_OK;
  }   
判断去掉也还是不行。可能是哪里的问题呢。

closed 2022-03-02 by liuning263