zlgopen/awtk #676: str_t字符串模块可否添加转换为GBK编码的API？

ufbycd, 2022-02-17:
国内市面上很多包含字库的打印模块所使用的编码是GBK的。
所以，如果str_t字符串模块有现成的转换为GBK编码的API将方便很多！

soyzhc, 2022-02-23:
可以看  awtk\src\charset 里面提供的api

xianjimli, 2022-02-24:
> 随便添加类似printf那样的格式化字符串API也很有用:
```
/**
 * @method str_format
 * 通过格式设置字符串。
 * @param {str_t*} str str对象。
 * @param {uint32_t} size format生成的字符串的最大长度(用于预习分配内存)。
 * @param {const char*} format 格式。
 *
 * @return {ret_t} 返回RET_OK表示成功，否则表示失败。
 */
ret_t str_format(str_t* str, uint32_t size, const char* format, ...);

/**
 * @method str_append_format
 * 通过格式追加字符串。
 * @param {str_t*} str str对象。
 * @param {uint32_t} size format生成的字符串的最大长度(用于预习分配内存)。
 * @param {const char*} format 格式。
 *
 * @return {ret_t} 返回RET_OK表示成功，否则表示失败。
 */
ret_t str_append_format(str_t* str, uint32_t size, const char* format, ...);
```

ufbycd, 2022-03-03:
> 可以看 awtk\src\charset 里面提供的api

哦，原来在这里。不错！

ufbycd, 2022-03-03:
> ```
> /**
>  * @method str_format
>  * 通过格式设置字符串。
>  * @param {str_t*} str str对象。
>  * @param {uint32_t} size format生成的字符串的最大长度(用于预习分配内存)。
>  * @param {const char*} format 格式。
>  *
>  * @return {ret_t} [...]

Well done！

ufbycd, 2022-03-03:
经测试awtk\src\charset里的API`encoding_utf8_to_gbk`发现，在X86_64 linux平台上转换成功、但在arm linuxa平台上的转换总失败。
怀疑是此平台的glibc编译的iconv支持有问题，于是在`encoding.c`文件的开关加入`#undef LINUX`以强制不使用iconv后字符串转换成功。
故不建议在linux平台默认使用iconv,而由用户选择是使用iconv还是awtk内部支持。