zlgopen/awtk #678: window_manager_switch_to 切换窗口窗口动画消失

hanzixi, 2022-02-18:
xml文件中对窗口打开和关闭做了动画效果,比如左右平移,在使用window_open打开窗口的时候有动画效果,使用window_close关闭窗口的的时候也有对应的关闭动画效果,但是使用window_close关闭之后该窗口控件之前设置的数据就会被清楚,再次打开的时候都又恢复到默认值了,不符合我们的需求;经查window_manager_switch_to 接口可以实现窗口切换且可以保留窗口数据,但是关闭窗口的动画又不会触发了,即在切换窗口的时候这个窗口绑定的动画不能被执行,目前正在查看源码寻找解决方案,目前无果,请问是否有较好的解决方案,谢谢;

xianjimli, 2022-02-18:
我这里测试没问题啊：

方法如下：

```
 ./bin/preview_ui design/default/ui/main_fscript.xml
```

* 1. 在main_fscript窗口点击“Basic"，打开"basic_fscript"窗口。

* 2.在basic_fscript窗口点击“Switch to"，打开“test_fscript”窗口。

* 3.在test_fscript窗口上的edit里输入123，再点击点击“Basic"，打开"basic_fscript"窗口。

* 4.在basic_fscript窗口点击“Switch to"，打开“test_fscript”窗口，可以看到之前输入的123存在，而且窗口动画正常。

你附加一个完整而简单的demo上来吧。

hanzixi, 2022-02-25:
李老师  抱歉回复晚了   我简单看了下demo  demo中使用的是open  等于重新打开了一个窗口;
下面的demo是我自己新建的一个简单的demo,一共三个页面,home_page,window1和window2;home_page一共有两个按钮,绑定的事件风别是打开window1和window2,window1和window2各有两个按钮,一个按钮是关闭当前窗口,另外一个是切换到home_page,使用的是window_manager_switch_to函数,测试发现使用该接口就没有了关闭该窗口的动画;而使用window_close就有动画;具体您可以看下下面这个简单的demo,谢谢您;
[window_test.zip](https://github.com/zlgopen/awtk/files/8139813/window_test.zip)
(如果下载无法解压,可以将后缀名修改为.rar)

xianjimli, 2022-02-28:
switch_to使用的是目标窗口的动画，你没有指定，所以没有。
close使用的当前窗口的动画。

hanzixi, 2022-03-01:
的确是的   homo_page没有设定动画;   如何动态的将其他窗口的关闭动画设置给home_page呢?能否通过widget_set_prop接口设置?设置完之后调用window_manager_switch_to会有动画效果?我们的需求就是关闭的窗口控件的数据得以保留,关闭窗口的时候要有相应的动画效果;另外就是window_open接口是每次都打开一个新的窗口,不管之前是否打开过,能否做到已经打开的窗口(后台运行)不再打开新的,而且打开之前已经打开过的?这样也可以节省内存;谢谢

xianjimli, 2022-03-01:
可以用widget_set_prop设置。

hanzixi, 2022-03-02:
李老师,通过widget_get_prop和widget_set_prop接口可以将窗口window1的动画设置给home_page,实现代码如下:
static ret_t on_switch_windows_click(void *ctx, event_t *e)
{
  widget_t *win = WIDGET(ctx);
  (void)ctx;
  (void)e;
  value_t v;
  bool_t open = FALSE;
  const char *anim_hint = NULL;
  const char *key = open ? WIDGET_PROP_OPEN_ANIM_HINT : WIDGET_PROP_CLOSE_ANIM_HINT;

  widget_t *home_win = widget_lookup(window_manager(), "home_page", TRUE);

  printf("win name: %s\r\n", win->name);
  if (home_win != NULL && win != NULL)
  {
    if (widget_get_prop(win, key, &v) == RET_OK)
    {
      anim_hint = value_str(&(v));
      printf("1win %s anit %s key %s\r\n", win->name, anim_hint, key);
    }
    else
    {
      key = WIDGET_PROP_ANIM_HINT;
      if (widget_get_prop(win, key, &v) == RET_OK)
      {
        anim_hint = value_str(&(v));
        printf("2win %s anit %s key %s\r\n", win->name, anim_hint, key);
      }
    }
    if (anim_hint != NULL)
    {
      window_t *home = WINDOW(win);
      widget_set_prop_str(home_win, key, anim_hint);
      widget_set_prop_str(home_win, WIDGET_PROP_OPEN_ANIM_HINT, anim_hint);
      printf("window %s anit state %d\r\n", home_win->name, home->window.disable_anim);
      widget_set_prop_bool(home_win, "reversed", TRUE);
    }
    return window_manager_switch_to(window_manager(), win, home_win, FALSE);
    
  }

  return RET_FAIL;
}
但是测试发现使用window_manager_switch_to接口只能播放打开窗口的动画,而无法播放关闭窗口的动画,查看代码发现该接口最终创建的是打开窗口的动画;手动修改关闭窗口可以实现关闭窗口的动画,但是该窗口也被关闭;
[image]

使用widget_set_prop_bool(home_win, "reversed", TRUE);将窗口动画反向,但是测试无效;
请问李老师,怎么有效的实现切换窗口播放的是关闭窗口动画并且不关闭该窗口呢;

下面是我修改过后的简单测试demo,麻烦您看下,看看能否修改,谢谢;

[window_test.zip](https://github.com/zlgopen/awtk/files/8168152/window_test.zip)

xianjimli, 2022-03-02:
怎么有效的实现切换窗口播放的是关闭窗口动画并且不关闭该窗口呢;
>不支持

hanzixi, 2022-03-03:
根据awtk源码,修改之后增加了自定义动画,实现切换窗口时候的反向动画,增加了open windows和switch windows的兼容处理,打开窗口的时候先查找下,如果没有打开这使用window_open接口,如果已经打开则使用switch_to接口,这样原先界面的数据还都存在;demo工程如下:
[window_test.zip](https://github.com/zlgopen/awtk/files/8177169/window_test.zip)

[image]

[image]

[image]