zlgopen/awtk #680: 树莓派编译awtk后无法正常运行

qylaoying, 2022-02-23:
我在Ubuntu下，按awtk说的办法下载安装相关软件后，能正常编译，按此办法，下载到树莓派（带桌面）后编译出错，随后下载安装了x11,并将Ubuntu下的glibconfig.h通过root拷到树莓派的/usr/include/glib-2.0/下， 编译通过，但运行./bin/demoui，报“段错误",用gdb调试发现demoui窗体会弹出，随后立即自动关闭并退出。为啥啊？树莓派下应该怎样才能正常编译？用的是最新安装的系统

qylaoying, 2022-02-23:
apt-get install xorg，我用这个命令装的x11，装后检查，发现/usr/include/X11下很多的文件，表明安装成功。https://bbs.csdn.net/topics/370161505是这里的办法

qylaoying, 2022-02-23:
或者告诉我你们的编译办法是什么，只要编译能成功，并且能正常运行

soyzhc, 2022-02-23:
默认编译使用opengles驱动，你的树莓派是否有开启显卡硬件加速？，具体可以通过sudo raspi-config命令配置

或者你试试配置awtk使用agge模式，要修改awtk_config.py文件，具体参考
https://awtk.zlg.cn/docs/AWTK/v1.7/AWTK%20Guide/1.%E7%AC%AC%E4%B8%80%E7%AB%A0%20%E5%88%9D%E8%AF%86AWTK.html#_1-8-1-%E6%B8%B2%E6%9F%93%E6%A8%A1%E5%BC%8F%E9%80%89%E6%8B%A9

soyzhc, 2022-02-23:
类似问题：https://github.com/zlgopen/awtk/issues/655

qylaoying, 2022-02-23:
使用agge模式，重新编译就好了，非常感谢