zlgopen/awtk #686: 问题咨询：移动view父控件，view中子控件不会自动调整大小和位置

Sunyhs, 2022-02-28:
一个view中存在一个slider，二个button，使用widget_move_resize（）函数移动view时，其中的slider和button仍在原地，有什么好方法可以实现子控件跟随父控件移动并对应调整大小

xianjimli, 2022-03-01:
位置是相对父控件的，应该会跟着移动的。只有设置了layout参数(比如百分比)，大小才会跟着调整。

Sunyhs, 2022-03-01:
<view name="view2" x="71%" y="10%" w="28%" h="36%">
    <view name="view_play" x="0" y="b" w="100%" h="57" style:normal:bg_color="#FFFFFF">
      <button name="button_play" x="0" y="m" w="6%" h="49%" style:normal:bg_color="#00000000" style:normal:icon_at="center" style:normal:bg_image_draw_type="icon"/>
      <slider name="slider" x="9%" y="m" w="68%" h="35%" style:normal:bg_color="#C7C7C7" style:over:bg_color="#C7C7C7" value="0"/>
      <label name="label_time" x="79%" y="m" w="6%" h="100%" style="left"/>
      <button name="button_close_full" x="r:2%" y="m" w="2%" h="59%" visible="false" style:normal:icon_at="center" style:normal:round_radius="0" enable="false"/>
    </view>
    <button name="button_full" x="r" y="b:16%" w="11%" h="14%" visible="true" style:normal:icon_at="center" style:normal:round_radius="0"/>
  </view>

Sunyhs, 2022-03-01:
``  <view name="view2" x="71%" y="10%" w="28%" h="36%">
    <view name="view_play" x="0" y="b" w="100%" h="57" style:normal:bg_color="#FFFFFF">
      <button name="button_play" x="0" y="m" w="6%" h="49%" style:normal:bg_color="#00000000" style:normal:icon_at="center" style:normal:bg_image_draw_type="icon"/>
      <slider name="slider" x="9%" y="m" w="68%" h="35%" style:normal:bg_color="#C7C7C7" style:over:bg_color="#C7C7C7" value="0"/>
      <label name="label_time" x="79%" y="m" w="6%" h="100%" style="left"/>
      <button name="button_close_full" x="r:2%" y="m" w="2%" h="59%" visible="false" style:normal:icon_at="center" style:normal:round_radius="0" enable="false"/>
    </view>
    <button name="button_full" x="r" y="b:16%" w="11%" h="14%" visible="true" style:normal:icon_at="center" style:normal:round_radius="0"/>
  </view>

Sunyhs, 2022-03-01:
**  <view name="view2" x="71%" y="10%" w="28%" h="36%">
    <view name="view_play" x="0" y="b" w="100%" h="57" style:normal:bg_color="#FFFFFF">
      <button name="button_play" x="0" y="m" w="6%" h="49%" style:normal:bg_color="#00000000" style:normal:icon_at="center" style:normal:bg_image_draw_type="icon"/>
      <slider name="slider" x="9%" y="m" w="68%" h="35%" style:normal:bg_color="#C7C7C7" style:over:bg_color="#C7C7C7" value="0"/>
      <label name="label_time" x="79%" y="m" w="6%" h="100%" style="left"/>
      <button name="button_close_full" x="r:2%" y="m" w="2%" h="59%" visible="false" style:normal:icon_at="center" style:normal:round_radius="0" enable="false"/>
    </view>
    <button name="button_full" x="r" y="b:16%" w="11%" h="14%" visible="true" style:normal:icon_at="center" style:normal:round_radius="0"/>
  </view>**

closed 2022-03-01 by Sunyhs

Sunyhs, 2022-03-01:
那个主要的问题是我在xml中设置百分比，当我用widget_move_resize（）时就破坏了原有的百分比构造（即布局），所以会出现view被移动，但是其子控件不会跟着移动。但是用widget_set_self_layout（）重新设置布局参数移动的话就不会破坏布局，子控件就会跟着父控件移动并调整。最初我以为使用像素设置和百分比设置只有xml中才有区别，现在看来如果想要保持对称关系的话，还是需要使用百分比（布局参数）来设计