zlgopen/awtk #687: 自定义控件实现拖动程序窗口移动的功能

18611429192, 2022-03-01:
我本来是想对系统自带的标题栏进行一下格式改造来适应程序风格的。但是awtk目前不支持，只能使用自定义控件来实现拖动程序的功能。

我使用`widget_get_native_window`函数获得程序窗口左上角的坐标`native_window->rect.x` 和, `native_window->rect.y`, 通过控件的移动事件，将移动的距离加到窗口左上角的坐标上，然后使用`native_window_move `函数来移动程序窗口。

结果是在windows下移动不能太快，太快了会导致鼠标跑到了程序外面，然后程序跟不上鼠标的移动，只能慢慢移动鼠标，窗口才能跟得上移动速度。

但是在linux下移动起来就没问题，很快的速度移动窗口也能跟得上。

**请问有没有方法能让windows下的窗口移动速度跟得上鼠标的速度？**

详情见视频
windows：

https://user-images.githubusercontent.com/49744753/156120046-9d8ea6cb-cc0f-4c50-a4d1-9b67574e4039.mp4

linux：

https://user-images.githubusercontent.com/49744753/156120106-7750f555-2e65-4868-a00f-073616a3ad13.mp4

18611429192, 2022-03-01:
大概代码如下
```
switch (type) {
        case EVT_POINTER_DOWN: {
            printf("down\n");
            title_bar->pressed = TRUE;
            pointer_event_t* pointer_event = (pointer_event_t*)e;
            printf("down:(%d, %d)\n", pointer_event->x, pointer_event->y);
            //获取当前点的全局坐标和本地坐标保存到结构体中
            title_bar->press_point_local.x = pointer_event->x;
            title_bar->press_point_local.y = pointer_event->y;
            title_bar->native_window = widget_get_native_window(title_bar);
            printf("native_window:(%d, %d),(%d,%d)\n", title_bar->native_window->rect.x, title_bar->native_window->rect.y, title_bar->native_window->rect.w, title_bar->native_window->rect.h);
            title_bar->press_point_screen.x = title_bar->native_window->rect.x;
            title_bar->press_point_screen.y = title_bar->native_window->rect.y;
            printf("press_point_screen:(%d, %d)\n", title_bar->press_point_screen.x, title_bar->press_point_screen.y);
            printf("press_point_local:(%d, %d)\n", title_bar->press_point_local.x, title_bar->press_point_local.y);
            break;
        }
        case EVT_POINTER_DOWN_ABORT: {
            printf("down_abort\n");
            if (title_bar->pressed) {
                printf("EVT_POINTER_DOWN_ABORT\n");
                title_bar_pointer_up_cleanup(widget);
            }
            break;
        }
        case EVT_POINTER_MOVE: {
            if (title_bar->pressed) {
                point_t* press_point_local = &(title_bar->press_point_local);
                printf("press_point_local:(%d, %d)\n", press_point_local->x, press_point_local->y);

                pointer_event_t* pointer_event = (pointer_event_t*)e;

                //根据之前记录的按下点的坐标，和当前点的坐标计算出要移动的距离，加到顶点坐标上，进行窗口移动
                title_bar->press_point_screen.x += (pointer_event->x - press_point_local->x);
                title_bar->press_point_screen.y += (pointer_event->y - press_point_local->y);

                printf("after cal p_vertex:(%d, %d)\n", title_bar->press_point_screen.x, title_bar->press_point_screen.y);

                native_window_move(title_bar->native_window, title_bar->press_point_screen.x, title_bar->press_point_screen.y, TRUE);
            }
            break;
        }
```

ufbycd, 2022-03-06:
AWTK在PC平台是用SDL库实现的，所以那个是SDL的标题栏，不如直接改SDL的标题栏。

18611429192, 2022-03-21:
能大概说下流程么？这个从来没接触过