zlgopen/awtk #694: slide_menu控件，能不能逐渐显示大小

wodingdong, 2022-03-04:
slide_menu控件，当前端显示有5个，最中间的第三个显示最大，第二个和第四个小一点，然后第一个和第五个更小一点？
[image]

现在看到的效果只是可以设置中间的那个稍大

xianjimli, 2022-03-04:
这个不太好弄

wodingdong, 2022-03-07:
> 这个不太好弄
这个我尝试重新设置旁边两个滑块的位置，设置完总是觉得看得别扭

wodingdong, 2022-03-07:
> 这个不太好弄

这个控件合理的设计应该还要能设计滑块的高度和宽度， 不能像现在长宽是一样的

wodingdong, 2022-03-07:
> 这个不太好弄

能不能添加一下支持

wodingdong, 2022-03-07:
> > 这个不太好弄
> > 这个我尝试重新设置旁边两个滑块的位置，设置完总是觉得看得别扭
[image]

wodingdong, 2022-07-04:
> 这个不太好弄
不太好弄的话，那能不能弄呢

wodingdong, 2022-07-08:
> ```c
> slide_menu_calc_child_size
> ```
好像并不行呢

wodingdong, 2022-07-08:
> ```c
> static int32_t slide_menu_calc_child_size(slide_menu_t* slide_menu, int32_t i, int32_t curr,
>                                           int32_t xo) {
>   float_t scale = 0;
>   int32_t max_size = WIDGET(slide_menu)->h;
>   float_t min_scale = slide_menu->min_scale;
>   int32_t xoffset = xo + max_size * (i - curr);
> 
>   if (tk_abs(xoffset) < max_size) {
>     scale = 1 + tk_abs(xoffset) * (min_scale - 1) / max_size;
>   } else {
>     scale = min_scale;
>   }
> 
>   return tk_roundi(max_size * scale);
> }
> ```
> 
> 改成下面的试试：
> 
> ```c
> static int32_t slide_menu_calc_child_size(slide_menu_t* slide_menu, int32_t i, int32_t curr,
>                                           int32_t xo) {
>   float_t scale = 0;
>   int32_t max_size = WIDGET(slide_menu)->h;
>   float_t min_scale = slide_menu->min_scale;
>   int32_t xoffset = xo + max_size * (i - curr);
> 
>    scale = 1 + tk_abs(xoffset) * (min_scale - 1) / max_size;
> 
>   return tk_roundi(max_size * scale);
> }
> ```

好像并不行呢

wodingdong, 2022-07-08:
> ```c
> static int32_t slide_menu_calc_child_size(slide_menu_t* slide_menu, int32_t i, int32_t curr,
>                                           int32_t xo) {
>   float_t scale = 0;
>   int32_t max_size = [...]

我想让slide_menu的块能逐渐显示大小，并且能设置块为长方形的，现在这些块只能设置成正方形

wodingdong, 2022-07-08:
> ```c
> static int32_t slide_menu_calc_child_size(slide_menu_t* slide_menu, int32_t i, int32_t curr,
>                                           int32_t xo) {
>   float_t scale = 0;
>   int32_t max_size = [...]

类似于这种：
[image]