zlgopen/awtk #697: 控件平移动画有残留片段

ufbycd, 2022-03-06:
使用下方描述的控件平移动画：
```xml
<view name="detect_card" x="0" y="0" w="100%" h="100%">
	<image x="c" y="200" w="24" h="145" image="card"
		animation="y(from=200, to=100, duration=2000, repeat_times=0, delay=500)" />
	<label x="c" y="80%" w="200" h="50" tr_text="please insert a reagent card" />
</view>
```
在x86_64 Linux平台测试正常，但在ARM Linux平台出现残留的控件片段：
[image]
x86_64 Linux平台正常

[image]
ARM Linux平台出现残留

xianjimli, 2022-03-07:
麻烦更新AWTK试试。谢谢

ufbycd, 2022-03-07:
嗯，问题解决了！Well done！

closed 2022-03-07 by ufbycd