zlgopen/awtk #700: 没有鼠标怎么使用输入法软键盘

Nagavesaka, 2022-03-09:
<edit name="search_text" x="right" y="10" w="80%" h="30" keyboard="kb_default_t9" focused="true"/>

我的linux设备只有GPIO扫键，怎么把焦点移到键盘上处理按键

xianjimli, 2022-03-09:
参考一下这两个文档：

https://github.com/zlgopen/awtk/blob/master/docs/widget_focus.md
https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_5keys_3keys.md

Nagavesaka, 2022-03-09:
> 参考一下这两个文档：
> 
> https://github.com/zlgopen/awtk/blob/master/docs/widget_focus.md https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_5keys_3keys.md

system_info_set_keyboard_type(system_info(), KEYBOARD_5KEYS);

不行哦，EDIT有焦点，但是软键盘拿不到焦点

Nagavesaka, 2022-03-09:
`<window name="search" bg_image="bg" bg_image_draw_type="scale"anim_hint="htranslate" >
<view x="left:10" y="10" w="60%" h="30" move_focus_prev_key="up" move_focus_next_key="down">
  <label x="left:10" y="10" w="20%" h="30" tr_text="Search"/>
  <edit name="search_text" x="right" y="10" w="80%" h="30" keyboard="kb_default_t9" focused="true"/>
</view>
  <list_view focusable="FALSE" name="" x="10" y="56" w="59%" h="45%" default_item_height="26">
    <scroll_view focusable="FALSE" name="scroll_view" x="0" y="0" w="-10" h="100%">
    
    </scroll_view>  
  </list_view>
</window>`

Nagavesaka, 2022-03-09:
<window name="search" bg_image="bg" bg_image_draw_type="scale"anim_hint="htranslate" >
<view x="left:10" y="10" w="60%" h="30" move_focus_prev_key="up" move_focus_next_key="down">
  <label x="left:10" y="10" w="20%" h="30" tr_text="Search"/>
  <edit name="search_text" x="right" y="10" w="80%" h="30" keyboard="kb_default_t9" focused="true"/>
</view>
  <list_view focusable="FALSE" name="" x="10" y="56" w="59%" h="45%" default_item_height="26">
    <scroll_view focusable="FALSE" name="scroll_view" x="0" y="0" w="-10" h="100%">
    
    </scroll_view>  
  </list_view>
</window>

Nagavesaka, 2022-03-09:
``<window name="search" bg_image="bg" bg_image_draw_type="scale"anim_hint="htranslate" >
<view x="left:10" y="10" w="60%" h="30" move_focus_prev_key="up" move_focus_next_key="down">
  <label x="left:10" y="10" w="20%" h="30" tr_text="Search"/>
  <edit name="search_text" x="right" y="10" w="80%" h="30" keyboard="kb_default_t9" focused="true"/>
</view>
  <list_view focusable="FALSE" name="" x="10" y="56" w="59%" h="45%" default_item_height="26">
    <scroll_view focusable="FALSE" name="scroll_view" x="0" y="0" w="-10" h="100%">
    
    </scroll_view>  
  </list_view>
</window>

Nagavesaka, 2022-03-10:
> 参考一下这两个文档：
> 
> https://github.com/zlgopen/awtk/blob/master/docs/widget_focus.md https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_5keys_3keys.md

而且，一旦EDIT失去焦点软件键盘就关闭了，这个有什么办法吗？

xianjimli, 2022-03-11:
不少朋友的产品都只有键盘，应该是可以的，看看文档。

```
  /** 
   * @property {bool_t} close_im_when_blured
   * @annotation ["set_prop","get_prop","readable","persitent","design","scriptable"]
   * 
   * 是否在失去焦点时关闭输入法(默认是)。
   *
   */
  bool_t close_im_when_blured;
```

RainMic1019, 2022-03-11:
> > 参考一下这两个文档：
> > https://github.com/zlgopen/awtk/blob/master/docs/widget_focus.md https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_5keys_3keys.md
> 
> system_info_set_keyboard_type(system_info(), KEYBOARD_5KEYS);
> 
> 不行哦，EDIT有焦点，但是软键盘拿不到焦点

我们这边测试过是可行的，在下面的文档中增加了详细步骤，你可以参考一下。
https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_5keys_3keys.md

Nagavesaka, 2022-03-12:
> 我们这边测试过是可行的，在下面的文档中增加了详细步骤，你可以参考一下。 https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_5keys_3keys.md

按照你的建议button可以拿到焦点了，现在  <candidates focusable="true" x="0" y="0" w="100%" h="30"/>还拿不到，麻烦再给点建议

RainMic1019, 2022-03-14:
> > > > 参考一下这两个文档：
> > > > https://github.com/zlgopen/awtk/blob/master/docs/widget_focus.md https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_5keys_3keys.md
> > > 
> > > 
> > > system_info_set_keyboard_type(system_info(), KEYBOARD_5KEYS);
> > > 不行哦，EDIT有焦点，但是软键盘拿不到焦点
> > 
> > 
> > 我们这边测试过是可行的，在下面的文档中增加了详细步骤，你可以参考一下。 https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_5keys_3keys.md
> 
> 按照你的建议button可以拿到焦点了，现在 candidates还拿不到，麻烦再给点建议

@Nagavesaka 将 candidates 控件的 focusable 属性设置为 true 就可以了，将所有你想进行焦点切换的控件的 focusable 属性都设置为 true。

Nagavesaka, 2022-03-14:
搞定了，之前candidates就有焦点了，但是颜色没配好，所以没看出来。

关闭这个ISSUE。感谢支持

closed 2022-03-14 by Nagavesaka