zlgopen/awtk #702: 问题咨询：切换语言后会触发edit注册的value_changed事件

Sunyhs, 2022-03-11:
为一个edit注册value_changed事件，切换语言后会自动调用该edit注册的事件回调函数，想问一下这是什么原因呢？

WNsACE, 2022-03-11:
你这个 edit 控件是不是用了 tr_text 属性来设置初值？方便把你的 ui 贴出来看看？

Sunyhs, 2022-03-12:
好想是这样的,每次切换语言下发的都是0
`<edit name="edit_num" w="20%" style="setting" x="25%" y="18%" h="5%" auto_fix="true" input_type="int" step="1"  tr_text="" min="1" max="750"/>`

Sunyhs, 2022-03-12:
> 你这个 edit 控件是不是用了 tr_text 属性来设置初值？方便把你的 ui 贴出来看看？
好想是这样的，这个会有影响是吧
`<edit name="edit_num" w="20%" style="default" x="25%" y="18%" h="5%" auto_fix="true" input_type="int" step="1" tr_text="" min="1" max="750"/>`

WNsACE, 2022-03-12:
是的，当切换语言的时候，就会去读取 tr_text 属性，然后根据 tr_text 修改 text 属性的，而 text 属性刚好是 edit 的 value 值，所以就会触发 value_changed 事件。
如果你没有用到 tr_text 的话，就删了吧