zlgopen/awtk #704: 问题咨询：如何触发自定义控件的on_paint_self函数

Sunyhs, 2022-03-14:
当点击一个button时 出弹出一个长条状的自定义控件，给自定义控件内部的 .on_paint_self写好自身的绘制函数， 绘制函数中需要根据不同的文本宽度来调整自定义控件的位置及大小 (widget_move_resize), 宽度使用widget_measure_text（）函数计算的得出，希望做到点击button时自适应宽调整，实际效果虽能达到，但是会出现宽度原来很长，将鼠标移入自定义控件范围内后（需要获得焦点？），又会根据文本的宽度自动调整，请问一下是什么原因？有什么办法可以在点击button以后就自适应而不是需要进一步操作才变化？是焦点的获取问题，还是自定义控件的on_paint_self的问题？

RainMic1019, 2022-03-14:
- 可以调用 widget_invalidate 接口重绘控件，如果需要强制重绘，则可以调用 widget_invalidate_force 接口。
- 鼠标移动到控件中，控件状态会由 normal 变为 over。

Sunyhs, 2022-03-14:
button控件的CLICK事件回调函数中会调用 widget_invalidate 接口重绘控件
代码已修改，如下
`  canvas_t* c = widget_get_canvas(widget));
  widget_on_paint_self（widge), c);  //先去触发一遍on_paint_self开启并设置初值
  widget_set_state(replay_bar->widget, WIDGET_STATE_OVER);   //将控件状态变为over
  widget_invalidate_force(widget, NULL);`

但实际的效果还是需要去动一动鼠标进去才行

Sunyhs, 2022-03-14:
button控件的CLICK事件回调函数中会调用 widget_invalidate 接口重绘控件
代码已修改，如下
`canvas_t* c = widget_get_canvas(widget)); 
widget_on_paint_self（widge), c); //先去触发一遍on_paint_self开启并设置初值 
widget_set_state(replay_bar->widget, WIDGET_STATE_OVER); //将控件状态变为over 
widget_invalidate_force(widget, NULL);`

但实际的效果还是需要去动一动鼠标进去才行

RainMic1019, 2022-03-14:
你可以再 widget_invalidate 函数和控件的 .on_paint_self 函数中打断点看看为什么鼠标不进入的时候坐标数据不对。

Sunyhs, 2022-03-15:
我这边通过widget_get_text_utf8（）函数拿到on_paint_self中的text文本发现是正确，但第一次执行paint_self函数时widget_measure_text（）计算的值为100， 当我移动鼠标时， 第二次执行paint_self函数，获取的文本未变，计算的宽度变为89

RainMic1019, 2022-03-21:
我这边测试时正常的，方便做个能重现的简单demo我们这边看看吗？