zlgopen/awtk #706: 只有按键环境，控件焦点切换的问题

Nagavesaka, 2022-03-18:
button之间焦点是可以用上下键切换焦点的，但是combox, switch无法键盘切换焦点，请问如何解决

RainMic1019, 2022-03-21:
1. 在AWTK Designer中创建的 combo_box_ex 是默认支持焦点停留的，可以切换焦点。
2. switch 控件勾选 focusable 属性（设置为true）后，也是支持焦点停留的，可以切换焦点。

如果你那边还是不行，能否发一下 UI 的 xml 文件来看看。

RainMic1019, 2022-03-21:
> button之间焦点是可以用上下键切换焦点的，但是combox, switch无法键盘切换焦点，请问如何解决

具体可以参考以下文档：

https://github.com/zlgopen/awtk/blob/master/docs/widget_focus.md
https://github.com/zlgopen/awtk/blob/master/docs/how_to_support_5keys_3keys.md

(awtk-c-demo)[https://github.com/zlgopen/awtk-c-demos] 的5keys示例中有 combo_box 和 switch 控件的示例，你可以看看。

Nagavesaka, 2022-03-21:
<window name="setting" bg_image="bg" bg_image_draw_type="scale" move_focus_up_key="up" move_focus_down_key="down" move_focus_left_key="left" move_focus_right_key="right">
  <view x="c" y="20" w="60%" h="120" children_layout="default(c=2,r=4,s=10)">
  <label style="right" focusable="false" name="language" tr_text="language"/>
  <combo_box name="language_selector" focused="true" focusable="true" readonly="true" options="1:English;2:中文;"/>
  <label style="right" focusable="false" name="key sound" tr_text="key sound"/>
  <switch style="default" name="key_sound" focusable="true" readonly="true"  tr_text="ON"/>
  <label style="right" focusable="false" tr_text="volumn"/>
  <slider style="default" name="volumn" value="60" min="1" max="100" step="3" focusable="true"/>
  <label style="right" focusable="false" name="theme" tr_text="theme"/>
  <combo_box name="theme_selector"  focusable="true" readonly="true" options="1:Default;2:Classic" />
  </view>
</window>

Nagavesaka, 2022-03-21:
`<window name="setting" bg_image="bg" bg_image_draw_type="scale" move_focus_up_key="up" move_focus_down_key="down" move_focus_left_key="left" move_focus_right_key="right">
  <view x="c" y="20" w="60%" h="120" children_layout="default(c=2,r=4,s=10)">
  <label style="right" focusable="false" name="language" tr_text="language"/>
  <combo_box name="language_selector" focused="true" focusable="true" readonly="true" options="1:English;2:中文;"/>
  <label style="right" focusable="false" name="key sound" tr_text="key sound"/>
  <switch style="default" name="key_sound" focusable="true" readonly="true"  tr_text="ON"/>
  <label style="right" focusable="false" tr_text="volumn"/>
  <slider style="default" name="volumn" value="60" min="1" max="100" step="3" focusable="true"/>
  <label style="right" focusable="false" name="theme" tr_text="theme"/>
  <combo_box name="theme_selector"  focusable="true" readonly="true" options="1:Default;2:Classic" />
  </view>
</window>
`

RainMic1019, 2022-03-22:
你可以参考 [awtk-c-demo](https://github.com/zlgopen/awtk-c-demos) 中的 5keys 示例，有类似的用法。