zlgopen/awtk #707: 只有按键的环境下输入键盘怎么把焦点交给其他控件

Nagavesaka, 2022-03-20:
<keyboard theme="keyboard" x="5" y="bottom:5" w="60%" h="40%" grab_keys="true">

当输入入法完成后，结果已显示在listview里了，但是我现在没办法把焦点给到listview里的list_item，并且用上下键进行选择

Nagavesaka, 2022-03-20:
经调试发现只有
input_method_t* im = input_method();
input_method_request(im, NULL);
关闭键盘后list_item才能拿到焦点，已设置edit_set_close_im_when_blured(widget, FALSE);
怎么样才能不关闭键盘还能让list_item拿到焦点呢？

RainMic1019, 2022-03-21:
> 经调试发现只有 input_method_t* im = input_method(); input_method_request(im, NULL); 关闭键盘后list_item才能拿到焦点，已设置edit_set_close_im_when_blured(widget, FALSE); 怎么样才能不关闭键盘还能让list_item拿到焦点呢？

目前不支持在软键盘打开的情况下让 list_item 拿到焦点，可以在输入法完成后将软键盘关闭，再将焦点从 edit 移动到 list_item 上，具体可以参考 (awtk-c-demo)[https://github.com/zlgopen/awtk-c-demos] 的5keys示例，里面新增了类似的用法。

RainMic1019, 2022-03-21:
[awtk-c-demo](https://github.com/zlgopen/awtk-c-demos)

closed 2022-03-21 by Nagavesaka