zlgopen/awtk #715: 关于AWTK License的问题

zenghaicang, 2022-04-12:
这个项目是基于LGPL-2.1 License。我的项目是基于STM32H7的单片机开发的，只能采用静态编译。MCU应用与实际应用场景相关，要开放项目源码，是不太现实的。我一直纠结于LVGL和AWTK之间，我想拥抱AWTK，也担心存在的风险，相信不少人也有这种顾虑。为了AWTK的成长，这个项目是否能修改成更加宽松的License呢？

xianjimli, 2022-04-12:
以前和老板确认过，AWTK基本功能绝不会收费，已经开源的代码绝不会闭源(我们也不想引起公愤，如果那样做，就相当于宣判AWTK的死刑)。

虽然用的LGPL，其实不管你怎么使用都可以，只有一个要求，如果修改了AWTK本身，我们有权利要求您开源修改部分的代码。目的是希望大家一起完善AWTK，防止一些公司私有化，维护自己的版本。

如果您修改了AWTK，也不想开放这部分代码或有法律顾虑，可以和我们签订免费商用合同，该合同中主要的要求是，我们有权在宣传时，提到贵公司和贵公司的产品用到了AWTK。

jserv, 2022-10-08:
> 虽然用的LGPL，其实不管你怎么使用都可以，只有一个要求，如果修改了AWTK本身，我们有权利要求您开源修改部分的代码。目的是希望大家一起完善AWTK，防止一些公司私有化，维护自己的版本。

是否考慮改用 [Mozilla Public License 2.0](https://opensource.org/licenses/MPL-2.0) (MPL-2.0) 呢？
若採用 MPL 的源碼經過修改後，必須也要採用 MPL 2.0 釋出。但是如果不是修改 MPL 的檔案，而是連結 (靜態與動態連結皆可) 或使用 MPL 的檔案，則是可以採用其他協議，也就是不需一定要開放源碼。