zlgopen/awtk #721: tk_istream_read_line()不能正确读取一行字符串

lvehe, 2022-04-18:
源码位置：awtk/src/tkc/istream.c
函数名称：int32_t tk_istream_read_line(...)

> if (p[offset] == '\n') {
>       break;

此处原意是想判断当前读取的字符是 '\n'，然后就跳出 while 循环。
但offset此时表示已经读取的字符的长度，所以p[offset]实际是定位到当前字符的下一个字符。
而此时，该字符却并没有获取到！所以此函数完全做不到读取以 '\n' 作为结束符的字符串。
如果用户之前未清空缓存p的话，读取的是p[offset]处的旧数据，如果恰好此旧数据是 '\n'，
则有可能导致错误地退出 while 循环。

修复起来也简单，只需正确定位当前索引即可，如下：
> if (p[offset - 1] == '\n') {
>       break;

xianjimli, 2022-04-18:
已修改，麻烦更新试试。非常感谢。

lvehe, 2022-04-19:
由于用的是串口读取以 '\n' 结尾的字符串，用新的代码后出现下面的错误：
tk_istream_eos:49 condition(stream->eos != NULL) failed!
经查，故障原因在于新的代码增加了如下判断：
```
if (read_bytes < 0) {
  break;
} else if (read_bytes == 0 && tk_istream_eos(stream)) {
  break;
}
```
上述的 tk_istream_eos(stream) 会失败，因为
awtk/src/streams/serial/istream_serial.c 中的
tk_istream_serial_create() 函数没有为eos赋值任何判断流结束的函数
```
  istream_serial->fd = fd;
  TK_ISTREAM(obj)->read = tk_istream_serial_read;
  TK_ISTREAM(obj)->flush = tk_istream_serial_flush;
  TK_ISTREAM(obj)->wait_for_data = tk_istream_serial_wait_for_data;

  return TK_ISTREAM(obj);
```

lvehe, 2022-04-19:
此外，
```
uint32_t start = 0;
uint32_t end = 0;  
...
start = time_now_ms();
end = start + timeout_ms;
```
start和end声明为uint32_t类型，与time_now_ms()的返回类型uint64_t不一致，可能出错。

xianjimli, 2022-04-19:
好的，谢谢。你再更新一下。