zlgopen/awtk #724: 是否可以在default.xml中动态改变字体大小

yuanyuanyuanquan, 2022-04-22:
目前我所有样式均在default.xml文件中进行实现，包括字体大小。现在我想设计一个大小可变化的样式，不知是否可以动态去改变default.xml中的值？？

xianjimli, 2022-04-22:
整体缩放字体：
```c
/**
 * @method system_info_set_font_scale
 * 设置字体缩放比例。
 *
 * @param {system_info_t*} info system_info对象。
 * @param {float_t} font_scale 字体缩放比例。
 *
 * @return {ret_t} 返回RET_OK表示成功，否则表示失败。
 */
ret_t system_info_set_font_scale(system_info_t* info, float_t font_scale);

```

yuanyuanyuanquan, 2022-06-24:
你好，李工，我最近遇到一个问题：
我们在测试过程中，发现在存在数据定时获取的页面上，跳转到自定义的弹窗界面时，会有该页面的功能函数无法触发的问题。或者还停留在原有的界面上在操作。
你这边遇到这样的问题吗？？

xianjimli, 2022-06-24:
可否弄个最简可重现的demo发给我看看。

yuanyuanyuanquan, 2022-06-24:
这边目前没有较合适的重现demo。但是大体重现的流程为：
进入具有定时数据加载的界面，按键进入到另一界面（该界面定时退出）操作后弹窗提示，弹窗为自定义弹窗（为全屏显示的普通弹窗，定时退出该界面，从而达到弹窗效果），在自定义弹窗的xml中有绑定任意键退出的函数。但是从此具有数据加载的页面执行此操作，按键退出函数完全没有触发。
有的界面数据加载频率高一点，就会呈现在该界面执行相应的按键函数，而当前显示界面相应按键函数未响应。