zlgopen/awtk #727: 进程占用CPU过高的问题

2297779320, 2022-04-26:
在hi3536上运行，双通道运行两个进程，单个进程CPU占用率浮动在16%~28%之间，总占用率在50%左右。有没有什么意见或者建议，降低下。

2297779320, 2022-04-27:
追看发现主要消耗函数是  Blit_RGB565_32 和   tk_memcpy16

xianjimli, 2022-04-27:
界面一直在画吗

2297779320, 2022-04-28:
我怀疑是的，我在一个窗口用了很多mutable_image这个控件，当我把这个窗口挂到后台 且隐藏 且不使能，他依然占用了很多CPU,我把这个窗口删掉就好了，但是我想保留它

2297779320, 2022-04-28:
补充，挂到后台后可以发现mutable_image的绘制回调并没有触发，但是依然占用了很多CPU，就很不能理解了

2297779320, 2022-04-29:
如果可以的话，希望能留一个群号给我，进群学习一下

xianjimli, 2022-05-06:
mutable_image的代码很少，自己调试一下吧。

xianjimli, 2022-05-06:
> 李工还望关注一下

> 如果可以的话，希望能留一个群号给我，进群学习一下

你可以加我的微信：xianjim，我拉你入群。

gcxfd, 2022-06-24:
chrome有个这个优化

[谷歌公布了Native Window Occlusion 新功能所带来的性能提升](https://www.linuxprobe.com/native-window-occlusion.html)
Chrome 现在会实时监控每个浏览器窗口的遮蔽状态，以降低 Windows 上完全遮蔽的窗口中打开的标签的优先级。

代码 https://bugs.chromium.org/p/chromium/issues/detail?id=1189281

据谷歌称，由于在 Windows 上启用了遮蔽检查功能，核心领域的性能得到了明显改善：

启动速度提高了 8.5% 至 25.8%。
GPU 显存使用量减少 3.1%。
渲染器绘制的帧数总体减少 20.4%。
遇到渲染器崩溃的客户减少 4.5%。
首次输入延迟降低 3.0%。
首次内容绘制和最大内容绘制的速度提高了 6.7%。
所有这些好处都要求 Chrome 用户在其设备上至少有两个 Chrome 浏览器窗口打开，而且其中一个窗口在桌面上完全被遮挡。