zlgopen/awtk #731: 怎么从window的name属性获取对应的UI文件名

archeng504, 2022-05-19:
打开窗口是用UI名打开的（如 /mainpage/mainpage.xml），但是window 的name属性是win_mainpage。判断窗口是否重复打开传的参数是UI名，没法判断呀。我看demo里面是直接将window 的name属性设成和UI xml的文件名一样。没有别的办法了吗？

WNsACE, 2022-05-19:
你是要想通过 widget 对象来获取窗口对于的资源名字？其实你要做什么呢？不太明白你的需求

archeng504, 2022-05-20:
就是用window_open打开新窗口时，我要判断window_manager中这个窗口是否已经打开了，已经打开了就不用重新打开，防止同一个窗口重复打开。window_manager只能用name属性来判断是否存在这个窗口，但是window_open 传进去的却是UI的资源名字，我就想枚举window_manager中的所有窗口，判断窗口的UI的资源名是否和要打开的窗口一致，一致的话就执行SwitchTo。

c++ 工程的TWindowFactory::Open 存在一个隐患，如果窗口UI的资源名字和窗口name属性不一致，就只会创建新的窗口，执行不了SwitchTo。如果反复打开相同的窗口，window_manager的窗口会不断的累加。

ret_t TWindowFactory::Open(const char* name, TAppWindow* caller, bool close_current,
                           TRequestPtr request) {
  char single_instance_prop_name[128];
  window_create_t create = NULL;
  bool_t single_instance = FALSE;
  return_value_if_fail(name != NULL, RET_BAD_PARAMS);
  gen_single_instance_prop_name(single_instance_prop_name, sizeof(single_instance_prop_name), name);

  create = (window_create_t)object_get_prop_pointer(mCreators, name);
  single_instance = object_get_prop_bool(mCreators, single_instance_prop_name, FALSE);
  return_value_if_fail(create != NULL, RET_NOT_FOUND);

  if (caller != NULL && single_instance && TAppWindow::isWindowOpen(name)) {
    return caller->SwitchTo(name, close_current, request);
  } else {
    TAppWindow* win = create(caller, close_current);
    return win->OnRequest(request, true);
  }
}

WNsACE, 2022-05-20:
1，正常来说，窗口打开的时候，把资源的名字设置为窗口名字（不管窗口是否有其他的名字），除非在打开后用户自己手动修改窗口名字。
2，窗口中有一个 single_instance 的属性用来控制窗口为单例，但是也是靠窗口名字来判断的，其逻辑就是把前一个相同名字的窗口关闭然后再打开一个新的窗口。

archeng504, 2022-05-20:
那只能把窗口的xml文件的name属性删掉或改成和xml同一个名字了

closed 2022-05-31 by archeng504